open62541/open62541 #3294: Event NodeIds

Muddasir-Shakil, 2019-11-21:
I was working with the Event tutorial. I was trying to set my own event NodeId but it was overwritten in UA_Server_createEvent. Further I looked in to the method that there is no check for already present NodeId for event node. I wanted to ask that if that is by design or a bug. Because in my application user could provide a NodeId (Numeric or String) for Event Node. If the NodeId is provided the server should work with the provided one, and should not generate a new one. If no NodeId provided than Server can work with the generated NodeId. 

version 1.1
OS linux

jpfr, 2019-11-21:
What is the use case here?
Note that it is an implementation detail of open62541 that we store events in nodes before triggering. Other sdks don’t do that...

sqadasi, 2019-11-22:
The current implementation of the  function `UA_Server_createEvent` don't have NodeId as input parameter. This function creates a Node with random and not used NodeId and stores it in the `outNodeId` parameter, so you could find that Node later on.

If you want to have your own NodeId, you have to create the Event Node using `add_ObjectNode ` function. Then you can use `UA_Server_triggerEvent` to trigger the Event.

Muddasir-Shakil, 2019-11-25:
> The current implementation of the function `UA_Server_createEvent` don't have NodeId as input parameter. This function creates a Node with random and not used NodeId and stores it in the `outNodeId` [...]

Thank you. This is a good trick. I will try this way.

paddor, 2022-03-10:
> The current implementation of the function `UA_Server_createEvent` don't have NodeId as input parameter. This function creates a Node with random and not used NodeId and stores it in the `outNodeId` parameter, so you could find that Node later on.

To what namespace does the new node ID belong to?

teccheck, 2025-02-04:
Seems to be abandoned. Feel free to reopen if needed.