open62541/open62541 #3295: howto read EUINFORMATION

tomtom5555, 2019-11-21:
Hi,
I'm working on an opc ua client Software (in Delphi Pascal!).
Reading endpoints, BrowseRequests, reading and writing variables and subscriptions are working well.
Now I stuck reading the Engineering Unit of a variable (EUINFORMATION).
Any hints how to do this are wellcome. Thanks a lot.
Tom

alrisback, 2021-09-02:
Hi!

Can you explain me how to use open62451 in Delphi?
Thanks Alessandro

tomtom5555, 2021-09-02:
Hi,
first, you have to translate all the constants, types and functions from c to pascal.
Then load the function adresses from the DLL.
Unfortunately, my code is embedded in a large project, so I can you give only some snipets.

Some example of types:
type
  TUA_Client=array[0..10000] of byte;
  PUA_Client =^TUA_Client;
  PUA_ClientConfig = ^TUA_ClientConfig;
  TUA_StatusCode = uint32;
  TUA_String = record
    len: size_t;
    data: PByte;
    function AsAnsiString: AnsiString;
    function AsString: String;
    procedure SetAnsiString(const a: AnsiString);
    procedure SetString(const a: string);
    procedure Free;
  end;
  PUA_String=^TUA_String;
  TUA_ByteString = TUA_String;
  PUA_ByteString = ^TUA_ByteString;
  TUA_LocalizedText = record
    locale: TUA_String;
    text: TUA_String;
  end;
  TUA_QualifiedName = record
    namespaceIndex: uint16;
    name: TUA_String;
  end;

  TUA_Guid = record
    data1: UInt32;
    data2: UInt16;
    data3: UInt16;
    data4: array[0..7] of byte;
  end;
  TUA_NodeId = record
    namespaceIndex: uint16;
    procedure Init;
    procedure SetNumericIdentifier(nsIndex: uint16; aidentifier: uint32);
    procedure SetStringIdentifier(nsIndex: uint16; aidentifier: string);
    procedure Assign(const aNodeId: TUA_NodeId);
    procedure WriteToStream(s: TStream);
    procedure ReadFromStream(s: TStream);
    procedure Free;
    case identifierType: int32 of
      UA_NODEIDTYPE_NUMERIC,
      UA_NODEIDTYPE_NUMERIC2,
      UA_NODEIDTYPE_NUMERIC4:   (numeric: uint32);
      UA_NODEIDTYPE_STRING:     (str: TUA_String);
      UA_NODEIDTYPE_GUID:       (guid: TUA_Guid);
      UA_NODEIDTYPE_BYTESTRING: (byteStr: TUA_ByteString);
  end;

The function prototypes:
var
  UA_TYPES: PUA_DataTypes;
  RootNode: TUA_NodeObj;

  UA_Client_new: function(): PUA_Client; cdecl;
  UA_ClientConfig_setDefault: function(config: PUA_ClientConfig): TUA_StatusCode; cdecl;
  UA_Client_getConfig: function(client: PUA_Client): PUA_ClientConfig; cdecl;
  UA_Client_connect: function(client: PUA_Client; const endpointUrl: PAnsiChar): TUA_StatusCode; cdecl;
  UA_Client_connect_username: function(client: PUA_Client; const endpointUrl, usr,pwd: PAnsiChar): TUA_StatusCode; cdecl;
  UA_Client_disconnect: function(client: PUA_Client): TUA_StatusCode; cdecl;
  UA_Client_delete: procedure(client: PUA_Client); cdecl;
  __UA_Client_writeAttribute: function(client: PUA_Client; var nodeId: TUA_NodeId;
              attributeId: TUA_AttributeId;  ain: pointer; atype: PUA_DataType): TUA_StatusCode; cdecl;
[... 38 lines cut ...]
      @UA_new                     := GetProcAddress(UA_Handle, 'UA_new');
      @__UA_Client_Service        := GetProcAddress(UA_Handle, '__UA_Client_Service');
      @UA_Variant_setScalarCopy   := GetProcAddress(UA_Handle, 'UA_Variant_setScalarCopy');
      @UA_Client_run_iterate      := GetProcAddress(UA_Handle, 'UA_Client_run_iterate');
      @UA_Client_Subscriptions_create  := GetProcAddress(UA_Handle, 'UA_Client_Subscriptions_create');
      @UA_Client_MonitoredItems_createDataChange  := GetProcAddress(UA_Handle, 'UA_Client_MonitoredItems_createDataChange');

      if @UA_Client_MonitoredItems_createDataChange=nil then begin
        FreeLibrary(UA_handle);
        UA_Handle:=0;
        FehlerText:='Error wrong OPC_UA DLL '+libopen62541;
      end;
    end else begin
      if UA_Handle<>0 then FreeLibrary(UA_handle);
      UA_Handle:=0;
      FehlerText:='Error loading OPC_UA DLL '+libopen62541;
    end;
  end;
  Result:=(UA_Handle<>0);
end;

Its a lot of typing work.
Hope this helps a little bit.

Tom

alrisback, 2021-09-02:
Dear Tom,
Thank you for reply. Do you have converted header file to pas file?

Br,Alessandro

tomtom5555, 2021-09-05:
Dear Alessandro,
sorry, but the converted headers are spread over a big project.
Take my examples as template to translate the functions you need.
Kind Regards 
Tom

alrisback, 2023-07-12:
Hi!

Do you have TUA_ClientConfig struct?

Tks

teccheck, 2025-02-04:
Seems to be abandoned. Feel free to reopen if needed.

closed 2025-02-04 by teccheck