open62541/open62541 #3299: Play mbedtls on stm32F429, handshake failed

zxb1717, 2019-11-27:
I'm afraid whether it is the right place to open this issue,if it is right here, my issue is like this, 

 Seeding the random number generator... ok
. Loading the CA root certificate ... ok (0 skipped)
. Connecting to tcp/qa2.greenlotstest.com/443...SSL: gethostbyname OK
Change DNS to IP:52.33.121.44:443
ok
. Setting up the SSL/TLS structure... ok
. Performing the SSL/TLS handshake...
--------------ssl->state = 0 -------------------
--------------ssl->state = 1 -------------------
--------------ssl->state = 2 -------------------
--------------ssl->state = 3 -------------------
--------------ssl->state = 4 -------------------
--------------ssl->state = 5 -------------------
--------------ssl->state = 6 -------------------
--------------ssl->state = 7 -------------------
--------------ssl->state = 8 -------------------
--------------ssl->state = 9 -------------------
--------------ssl->state = 10 -------------------
--------------ssl->state = 11 -------------------
--------------ssl->state = 17 -------------------
..\PolarSSL\src\ssl_tls.c:4095: is a fatal alert message (msg 40)
..\PolarSSL\src\ssl_tls.c:3822: mbedtls_ssl_handle_message_type() returned -30592 (-0x7780)
..\PolarSSL\src\ssl_cli.c:3279: mbedtls_ssl_read_record() returned -30592 (-0x7780)
........

the value of ssl->state is defined in ssl.h:
typedef enum
{
MBEDTLS_SSL_HELLO_REQUEST,	         //=0
MBEDTLS_SSL_CLIENT_HELLO,	           //=1
MBEDTLS_SSL_SERVER_HELLO,	            //=2
MBEDTLS_SSL_SERVER_CERTIFICATE,	      //=3
MBEDTLS_SSL_SERVER_KEY_EXCHANGE,	//=4
MBEDTLS_SSL_CERTIFICATE_REQUEST,	      //=5
MBEDTLS_SSL_SERVER_HELLO_DONE,	     //=6
MBEDTLS_SSL_CLIENT_CERTIFICATE,	     //=7
MBEDTLS_SSL_CLIENT_KEY_EXCHANGE,	     //=8
MBEDTLS_SSL_CERTIFICATE_VERIFY,	     //=9
MBEDTLS_SSL_CLIENT_CHANGE_CIPHER_SPEC,	//=10
MBEDTLS_SSL_CLIENT_FINISHED,	           //=11
MBEDTLS_SSL_SERVER_CHANGE_CIPHER_SPEC,	//=12
MBEDTLS_SSL_SERVER_FINISHED,	          //=13
MBEDTLS_SSL_FLUSH_BUFFERS,	         //=14
MBEDTLS_SSL_HANDSHAKE_WRAPUP,	//=15
MBEDTLS_SSL_HANDSHAKE_OVER,	//=16
MBEDTLS_SSL_SERVER_NEW_SESSION_TICKET,	//=17
MBEDTLS_SSL_SERVER_HELLO_VERIFY_REQUEST_SENT,//=18
}
mbedtls_ssl_states;
this is my log above, i tried many times,and it always stucked here at the same place and got the same code. someone can give me any suggestion, thanks. if more detailed log needed, i will upload it.

ChHaa, 2019-11-28:
Wrong place to ask as you are trying to perform a TLS handshake.

zxb1717, 2019-11-29:
@ChHaa Sorry

closed 2019-11-29 by zxb1717