open62541/open62541 #3301: Proposal: Use Ninja for CI

mlgiraud, 2019-11-27:
I've successfully sped up the build process on my local machine by about 30 seconds when building with all options enabled by using ninja instead of make.

Maybe we could speed up the CI process a bit, by switching to ninja as the default generator for the CI tests? Although this would only work, if ninja doesn't have to be installed as dependency every time (is it possible to cache a dependency installation?)

What do you think @Pro ?

Pro, 2019-11-27:
I'd say we should at least run a subset of CI with make to make sure that is always building, since most of the users will use GNU Makefiles.

For the rest we can definitely switch to ninja. If you want to propose a PR, feel free.

jpfr, 2019-11-29:
Hey @mlgiraud ,
ninja aggressively parallelizes.
Did you compare with `make -j` parallel make?

mlgiraud, 2019-11-29:
I used make with -j and -j 8 (on a 4 core system with 8 hyper cores) with these results

closed 2025-02-13 by marwinglaser