open62541/open62541 #3303: IP address

magicwolf2019, 2019-11-28:
Can I get an IP address when opening a session on the server?

jpfr, 2019-11-29:
Not with the normal API.
You can use *internal API* to get the IP address.

The activateSession callback gives you the session id.
https://github.com/open62541/open62541/blob/master/include/open62541/plugin/accesscontrol.h#L39

Then you can use internal API to retrieve the session.
The session has a pointer to the SecureChannel.
The SecureChannel points to the UA_Connection.
The UA_Connection has the socket.
On Linux, getpeername gives you the remote name or ip address.

magicwolf2019, 2019-12-02:
Thanks for the answer!
There is no session pointer in the callback function activateSession. There is only a pointer to "const UA_NodeId *sessionId". Please tell me how you can get a pointer to the SecureChannel using this sessionId?

magicwolf2019, 2019-12-04:
Thanks again for the answer! I understood how to do it. As a result, I made the following changes to several functions of SDK.
1. In order to save IP address information - I added one variable in the structure UA_Session and UA_Connection:
`    ...
	UA_SecureChannel *channel;
} UA_Session;

struct UA_Connection {
	...
    UA_UInt64 connectCallbackID;   /* Callback Id, for the connect-loop */
    struct sockaddr_storage remote_ip;
`
2. Saved ip address information and channel:
`static UA_StatusCode
ServerNetworkLayerTCP_add(UA_ServerNetworkLayer *nl, ServerNetworkLayerTCP *layer,
                          UA_Int32 newsockfd, struct sockaddr_storage *remote) {
	...
    c->openingDate = UA_DateTime_nowMonotonic();
    memcpy(&c->remote_ip, remote, sizeof(struct sockaddr_storage));
	
/* Creates and adds a session. But it is not yet attached to a secure channel. */
UA_StatusCode
UA_SessionManager_createSession(UA_SessionManager *sm, UA_SecureChannel *channel,
                                const UA_CreateSessionRequest *request, UA_Session **session) {
	...
    newentry->session.header.authenticationToken = UA_NODEID_GUID(1, UA_Guid_random());
    newentry->session.channel = channel;
`
3. I changed the function activateSession_default a bit to get information about IP and user in it.
`
typedef struct {
  struct sockaddr_storage remote_ip;
  int num_user;
} UA_SessionContext_st;

static UA_StatusCode
activateSession_default(UA_Server *server, UA_AccessControl *ac,
                        const UA_EndpointDescription *endpointDescription,
                        const UA_ByteString *secureChannelRemoteCertificate,
                        const UA_NodeId *sessionId,
                        const UA_ExtensionObject *userIdentityToken,
                        void **sessionContext) {
    AccessControlContext *context = (AccessControlContext*)ac->context;
    UA_Session *session = UA_SessionManager_getSessionById(&server->sessionManager, sessionId);
    UA_SecureChannel * channel=session->channel;
    UA_Connection * connection = channel->connection;
    SessionContext_st * sc;
    
    sc = (SessionContext_st *)*sessionContext;
    if (sc) {
      sc->num_user = -1; // anonymous
      memcpy(&sc->remote_ip, &connection->remote_ip, sizeof(struct sockaddr_storage));
    }

    /* The empty token is interpreted as anonymous */
    if(userIdentityToken->encoding == UA_EXTENSIONOBJECT_ENCODED_NOBODY) {
        if(!context->allowAnonymous)
            return UA_STATUSCODE_BADIDENTITYTOKENINVALID;

        /* No userdata atm */
        //*sessionContext = NULL;
        return UA_STATUSCODE_GOOD;
    }

    /* Could the token be decoded? */
    if(userIdentityToken->encoding < UA_EXTENSIONOBJECT_DECODED)
        return UA_STATUSCODE_BADIDENTITYTOKENINVALID;

    /* Anonymous login */
    if(userIdentityToken->content.decoded.type == &UA_TYPES[UA_TYPES_ANONYMOUSIDENTITYTOKEN]) {
        if(!context->allowAnonymous)
[... 80 lines cut ...]
    sc=malloc(sizeof(UA_SessionContext_st));
    *sessionContext=sc;

    retval = activateSession_old(server, ac, endpointDescription, secureChannelRemoteCertificate, sessionId, userIdentityToken, sessionContext);
    if ( retval != UA_STATUSCODE_GOOD )
      return retval;
    sc=*sessionContext;
    getnameinfo((struct sockaddr*)&sc->remote_ip,sizeof(struct sockaddr_storage), ip_str, sizeof(ip_str), NULL, 0, NI_NUMERICHOST);
    if ( client_check(ip_str,sc)==0 )
      return UA_STATUSCODE_GOOD;
    else
      return UA_STATUSCODE_BADUSERACCESSDENIED;
}
`
3. I also created my function closeSession:
`
static void
closeSession_my(UA_Server *server, UA_AccessControl *ac,
                     const UA_NodeId *sessionId, void *sessionContext) {
    /* no context to clean up */
  if (sessionContext != 0)
    free(sessionContext);
}
`
4. Function client_check performs the role simple firewall.

jpfr, 2019-12-09:
Nice!
To save this solution for posterity, I added a reference to the Wiki.

https://github.com/open62541/open62541/wiki#faq

closed 2019-12-09 by jpfr

magicwolf2019, 2019-12-09:
Thanks! Is it possible to add at least the first two changes to the sdk source code? In order not to edit the source code of the sdk every time you download from github. The third change is also important, but not universal ;(.

mpfj, 2021-03-03:
Sorry to drag this old issue up from the past but I've just spent a while exploring the same problem.

It would be a nice feature to add this as a default

mpfj, 2021-03-07:
Just to add to this, the above solution **isn't** possible if your using the "amalgamated code" option.

"open62541.h" doesn't include the necessary function and struct declarations.
They are all embedded inside "open62541.c" so the following code...

```c
// get client ip data
UA_AccessControl* context = (UA_AccessControl*)ac->context;
UA_Session* session = UA_Server_getSessionById(server, sessionId);
UA_SecureChannel* channel = session->channel;
UA_Connection* connection = channel->connection;
```

... gets the following errors...

```
opcua.cpp: In function 'UA_StatusCode activateSession(UA_Server*, UA_AccessControl*, const UA_EndpointDescription*, const UA_ByteString*, const UA_NodeId*, const UA_ExtensionObject*, void**)':
opcua.cpp:1222:4: error: 'UA_Session' was not declared in this scope
opcua.cpp:1222:16: error: 'session' was not declared in this scope
opcua.cpp:1222:68: error: 'UA_Server_getSessionById' was not declared in this scope
opcua.cpp:1224:39: error: invalid use of incomplete type 'UA_SecureChannel {aka struct UA_SecureChannel}'
./open62541.h:23615:8: error: forward declaration of 'UA_SecureChannel {aka struct UA_SecureChannel}'
```

magicwolf2019, 2021-03-07:
In the case UA_ENABLE_AMALGAMATION=OFF, then you need to include the appropriate header files.
I also added to the function activateSession_default receiving policy and application name:
`typedef struct {
  struct sockaddr_storage remote_ip;
  int num_user;
  char applicationName[101];
  char policyUri[101];
} UA_SessionContext_st;

static UA_StatusCode
activateSession_default(UA_Server *server, UA_AccessControl *ac,
                        const UA_EndpointDescription *endpointDescription,
                        const UA_ByteString *secureChannelRemoteCertificate,
                        const UA_NodeId *sessionId,
                        const UA_ExtensionObject *userIdentityToken,
                        void **sessionContext) {
    AccessControlContext *context = (AccessControlContext*)ac->context;
    UA_Session *session = UA_Server_getSessionById(server, sessionId);
    UA_SecureChannel * channel=0;
    UA_Connection * connection=0;
    if ( session!=0 ) {
      channel=session->channel;
      if ( channel!=0 && channel->state!=UA_SECURECHANNELSTATE_CLOSED && channel->state!=UA_SECURECHANNELSTATE_CLOSING )
        connection = channel->connection;
    }
    UA_SessionContext_st * sc;
    int len;

    sc = (UA_SessionContext_st *)*sessionContext;
    if (sc) {
      sc->num_user = -1; // anonymous
      if ( connection!=0 )
        memcpy(&sc->remote_ip, &connection->remote_ip, sizeof(struct sockaddr_storage));
      else
        memset(&sc->remote_ip, 0, sizeof(struct sockaddr_storage));
      memset(sc->applicationName,0,101); sc->applicationName[0]='?';
      if ( session!=0 ) {
        len=session->clientDescription.applicationName.text.length;
        if ( len>1 ) {
          if ( len>100 ) len=100;
          memcpy(sc->applicationName,session->clientDescription.applicationName.text.data,len);
        }
      }
      memset(sc->policyUri,0,101); sc->policyUri[0]='?';
      if ( channel!=0 ) {
        len=channel->securityPolicy->policyUri.length;
        if ( len>1 ) {
          char * policy=strchr(channel->securityPolicy->policyUri.data,'#');
          if ( policy!=0 )
            len=policy-channel->securityPolicy->policyUri.data;
          else
            policy=channel->securityPolicy->policyUri.data;
          if ( len>100 ) len=100;
          memcpy(sc->policyUri,policy,len);
        }
      }
    }
    /* The empty token is interpreted as anonymous */
    if(userIdentityToken->encoding == UA_EXTENSIONOBJECT_ENCODED_NOBODY) {
        if(!context->allowAnonymous)
            return UA_STATUSCODE_BADIDENTITYTOKENINVALID;

        /* No userdata atm */
        //*sessionContext = NULL;
        return UA_STATUSCODE_GOOD;
    }

    /* Could the token be decoded? */
    if(userIdentityToken->encoding < UA_EXTENSIONOBJECT_DECODED)
        return UA_STATUSCODE_BADIDENTITYTOKENINVALID;

    /* Anonymous login */
    if(userIdentityToken->content.decoded.type == &UA_TYPES[UA_TYPES_ANONYMOUSIDENTITYTOKEN]) {
        if(!context->allowAnonymous)
            return UA_STATUSCODE_BADIDENTITYTOKENINVALID;

        const UA_AnonymousIdentityToken *token = (UA_AnonymousIdentityToken*)
            userIdentityToken->content.decoded.data;

        /* Compatibility notice: Siemens OPC Scout v10 provides an empty
         * policyId. This is not compliant. For compatibility, assume that empty
         * policyId == ANONYMOUS_POLICY */
        if(token->policyId.data && !UA_String_equal(&token->policyId, &anonymous_policy))
            return UA_STATUSCODE_BADIDENTITYTOKENINVALID;

        /* No userdata atm */
        //*sessionContext = NULL;
        return UA_STATUSCODE_GOOD;
    }

    /* Username and password */
    if(userIdentityToken->content.decoded.type == &UA_TYPES[UA_TYPES_USERNAMEIDENTITYTOKEN]) {
        const UA_UserNameIdentityToken *userToken =
            (UA_UserNameIdentityToken*)userIdentityToken->content.decoded.data;

        if(!UA_String_equal(&userToken->policyId, &username_policy))
            return UA_STATUSCODE_BADIDENTITYTOKENINVALID;

        /* The userToken has been decrypted by the server before forwarding
         * it to the plugin. This information can be used here. */
        /* if(userToken->encryptionAlgorithm.length > 0) {} */

        /* Empty username and password */
        if(userToken->userName.length == 0 && userToken->password.length == 0)
            return UA_STATUSCODE_BADIDENTITYTOKENINVALID;

        /* Try to match username/pw */
        UA_Boolean match = false;
        for(size_t i = 0; i < context->usernamePasswordLoginSize; i++) {
            if(UA_String_equal(&userToken->userName, &context->usernamePasswordLogin[i].username) &&
               UA_String_equal(&userToken->password, &context->usernamePasswordLogin[i].password)) {
                match = true;
                if (sc)
                  sc->num_user = i;
                break;
            }
        }
        if ( context->allowAnonymous!=0 )
          match=true;
        if(!match)
            return UA_STATUSCODE_BADUSERACCESSDENIED;

        /* No userdata atm */
        //*sessionContext = NULL;
        return UA_STATUSCODE_GOOD;
    }

    /* Unsupported token type */
    return UA_STATUSCODE_BADIDENTITYTOKENINVALID;
}
`
And replaced the definition:
`{UA_STATUSCODE_BADIDENTITYTOKENINVALID, "BadUserIdentityTokenInvalid"},`

mpfj, 2021-03-07:
> In the case UA_ENABLE_AMALGAMATION=OFF, then you need to include the appropriate header files before function activateSession_default.

Of course, but I'm cross-compiling to an ARM platform so needed to use UA_ENABLE_AMALGAMATION=**ON**
I then (in theory) only need to #include "open62541.h" rather than all the individual #include files.

But the required structure definitions are not included in "open62541.h" and therefore can't be accessed with any of my code.

I guess the only option is to see how to move the structs / etc into the amalgamated header file somehow???

magicwolf2019, 2021-03-07:
Then you have problem creation phase on the files open62541.c and open62541.h. When the definition is used UA_ENABLE_AMALGAMATION:  Compile a single-file release into the files open62541.c and open62541.h.
https://open62541.org/doc/current/building.html
Maybe you do not happen? Do you have files open62541.c and open62541.h? They must have all the functions and definitions.

magicwolf2019, 2021-03-07:
I compile under Windows:
https://stackoverflow.com/questions/54985518/build-open62541-dll-with-visual-studio-2017
After launch:
cmake .. -G "Visual Studio 14 2015" -DUA_ENABLE_ENCRYPTION=ON -DUA_ENABLE_AMALGAMATION=ON ...
I get two files open62541.c and open62541.h. But before that I make changes to files:
src\server\ua_session.h
include\open62541\plugin\network.h
arch\network_tcp.c
src\server\ua_services_session.c
plugins\ua_accesscontrol_default.c

KittenWo0f, 2025-01-28:
I was building a library for my project according to [these](https://github.com/open62541/open62541/issues/3303#issuecomment-561571534) instructions in version 1.3.8. Everything works fine. 

Now I need to update the library to the current version 1.4.9 to use the new functions.
I found out that the `ServerNetworkLayerTCP_add` function and `UA_Connection` structure are missing in version 1.4.9. Doing some research on my own has not yielded any results. 

Is there a current way to get the ip address of the connecting client?

magicwolf2019, 2025-01-29:
I am not using version 1.4.9. But I will try to change this solution for the new version, but it will take me some time...

magicwolf2019, 2025-01-29:
The UA_getnameinfo function is currently being called in TCP_listenSocketCallback. And in this function, it seems to me, this problem is solved in a more elegant way:
`    /* Forward the remote hostname to the application */
    UA_KeyValuePair kvp;
    kvp.key = UA_QUALIFIEDNAME(0, "remote-address");
    UA_String hostName = UA_STRING(hoststr);
    UA_Variant_setScalar(&kvp.value, &hostName, &UA_TYPES[UA_TYPES_STRING]);
`
You can see that you can get hoststr (host_name) via "remote-address". But I can't say how to do it, have to try.
Most likely, you don't need to edit anything else in the code!

magicwolf2019, 2025-01-29:
Check the "MulticastDiscoveryCallback" function code:
`    const UA_String *address = (const UA_String*)
        UA_KeyValueMap_getScalar(params, UA_QUALIFIEDNAME(0, "remote-address"),
                                 &UA_TYPES[UA_TYPES_STRING]);
`

KittenWo0f, 2025-01-29:
Thanks for the quick response. 

I looked at getting the “remote-address” attribute in `MulticastDiscoveryCallback`. It gets this attribute from `const UA_KeyValueMap *params`. As far as I understand it is an internal part of the library and how to get `UA_KeyValueMap *params` in the user code is not clear to me.

In the [documentation](https://www.open62541.org/doc/master/plugin_eventloop.html#tcp-connection-manager) I found that the attribute “remote-address” can be obtained through “Active Connection Connection Callback”. I can't understand if this can be used in this case and if this callback can be set somewhere.

magicwolf2019, 2025-01-30:
Unfortunately, I agree with you - how to get the "remote-address" value is unclear ;(. It is also unclear how to activate the "ConnectionCallback".

magicwolf2019, 2025-01-30:
I found that it is possible to get the client's computer name like this:
```
#include <stdio.h>
#include <open62541/server.h>

UA_StatusCode (*activateSession_default)(UA_Server *server, UA_AccessControl *ac,
                                  const UA_EndpointDescription *endpointDescription,
                                  const UA_ByteString *secureChannelRemoteCertificate,
                                  const UA_NodeId *sessionId,
                                  const UA_ExtensionObject *userIdentityToken,
                                  void **sessionContext);

static UA_StatusCode
activateSession_my(UA_Server *server, UA_AccessControl *ac,
                        const UA_EndpointDescription *endpointDescription,
                        const UA_ByteString *secureChannelRemoteCertificate,
                        const UA_NodeId *sessionId,
                        const UA_ExtensionObject *userIdentityToken,
                        void **sessionContext) {
    UA_StatusCode retval;
    printf("activateSession_my:\n");
    if ( activateSession_default==0 )
      return UA_STATUSCODE_GOOD;
    retval = activateSession_default(server, ac, endpointDescription, secureChannelRemoteCertificate, sessionId, userIdentityToken, sessionContext);

    UA_QualifiedName key_clientDescription = {0, UA_STRING_STATIC("clientDescription")};
    UA_Variant clientDescription;
    UA_Server_getSessionAttribute(server, sessionId, key_clientDescription, &clientDescription);
    UA_ApplicationDescription * description=(UA_ApplicationDescription *)clientDescription.data;

        printf("\n\tName: %.*s", (int) description->applicationName.text.length,
               description->applicationName.text.data);
        printf("\n\tApplication URI: %.*s", (int) description->applicationUri.length,
               description->applicationUri.data);
        printf("\n\tProduct URI: %.*s", (int) description->productUri.length,
               description->productUri.data);
        printf("\n\tType: ");
        switch(description->applicationType) {
            case UA_APPLICATIONTYPE_SERVER:
                printf("Server");
                break;
            case UA_APPLICATIONTYPE_CLIENT:
                printf("Client");
                break;
            case UA_APPLICATIONTYPE_CLIENTANDSERVER:
                printf("Client and Server");
                break;
            case UA_APPLICATIONTYPE_DISCOVERYSERVER:
                printf("Discovery Server");
                break;
            default:
                printf("Unknown");
        }
        printf("\n\tDiscovery URLs:");
        for(size_t j = 0; j < description->discoveryUrlsSize; j++) {
            printf("\n\t\t[%lu]: %.*s", (unsigned long) j,
                   (int) description->discoveryUrls[j].length,
                   description->discoveryUrls[j].data);
        }
        printf("\n\n");

    return retval; 
}

static void
closeSession_my(UA_Server *server, UA_AccessControl *ac,
                     const UA_NodeId *sessionId, void *sessionContext) {
    printf("closeSession_arm:\n");
}

int main(void) {
    UA_Server *server = UA_Server_new();
    UA_ServerConfig *config = UA_Server_getConfig(server);

  activateSession_default=config->accessControl.activateSession;
  config->accessControl.activateSession = activateSession_my;
  config->accessControl.closeSession = closeSession_my;

    printf("start...\n");
    UA_Server_runUntilInterrupt(server);
    UA_Server_delete(server);
    return 0;
}

```

maslankam, 2025-09-10:
Thank you for responce. Do you know any examples of implementation of EventLoop on client side? I'm not sure if I understand all the concepts right because my previous implementation of eventloop didn't work.