open62541/open62541 #3304: Callback to read variable value from OPC UA Server

yokananwis, 2019-11-29:
Currently, I build a protocol converter from modbus TCP to OPC UA (as server) in beaglebone with ADA language accessing this library (in C). OPC UA server has already running and can convert modbus data to opc variables by polling modbus registers. The problem is when converting data when OPC UA Client (simulator) give command (change variable's value) to OPC UA Server (beaglebone), OPC UA Server can't get the changed value from client? Is there any callback/function to get the variable's value when the client give commands?

This is my configuration:

 - open62541 Version 1.1.0-dev-109:
 - operating system: Debian Stretch

jpfr, 2019-11-29:
Of course this is possible.

See these links:
https://open62541.org/doc/current/server.html#data-source-callback
https://open62541.org/doc/current/server.html#value-callback

yokananwis, 2019-11-30:
Thanks @jpfr for the response.

I have try these 2 callbacks, data source callback and value callback.

The OPC Server's Variable is changed either because of OPC client request or when I insert the value directly into it (when there is modbus data that should be converted to OPC UA variables) and both of these situation triggers the data source and value callback.

All I want is the callback that only triggered by OPC client request. Is it possible? Or can I filter the request from OPC client only?

jonathan-ibex-controls, 2023-03-01:
Does anyone know how to do this? The above links are broken. I have a working callback function, but the arguments have lots of layers of indirection and I can't find any documentation for how to use them.

 `tankLevelChangeNotificationCallback(UA_Server* server, UA_UInt32 monitoredItemId,
    void* monitoredItemContext, const UA_NodeId* nodeId,
    void* nodeContext, UA_UInt32 attributeId,
    const UA_DataValue* value)`

I tried printing value using the UA_Print function, but it just prints out "{ }":
`    UA_String out = UA_STRING_NULL;
        UA_print(&value, &UA_TYPES[UA_TYPES_DATAVALUE], &out);

        printf("%.*s\n", (int)out.length, out.data);
        UA_String_clear(&out);`

neojaw, 2023-03-01:
The above links should point to
https://www.open62541.org/doc/master/server.html#data-source-callback
https://www.open62541.org/doc/master/plugin_nodestore.html#value-callback
but I think none of them is what you are looking for, you are dealing with [local monitored items](https://www.open62541.org/doc/master/server.html#local-monitoreditems). I haven't worked with them, but if they are anything like monitored items on client side, the data is in `value->value`.

Also, your `UA_print` should take `value`, not `&value`, since it is already a pointer.

neojaw, 2023-03-01:
While this issue is up, I ran into the same problem as @yokananwis, needing to distinguish between external and internal writes to variables. Since the server uses the admin session to write variables, I figured I could check if the current session is the admin session:
```
void variable_write_callback(UA_Server *server, const UA_NodeId *session_id, void *session_context, const UA_NodeId *node_id, void *node_context, const UA_NumericRange *range, const UA_DataValue *data){
	if(session_id->identifier.guid.data1 == 1){
		// internal write
	}else{
		// external write
	}
}
```
It works, but I don't like that I have to hard code the session id but I cannot directly access the (internal) admin session object. Is there another, better way of doing this? (I'm on the 1.3.x branch if that is relevant.)

c-logic, 2024-02-15:
thanks neojaw :) it's also interesting for reading in read-callback

static void beforeReadTime(UA_Server *server, const UA_NodeId *sessionId, void *sessionContext, const UA_NodeId *nodeid, void *nodeContext, const UA_NumericRange *range, const UA_DataValue *data) {
    if(sessionId->identifier.guid.data1==1) //prevent internal looping
        return;
    UA_Variant xxx;
    UA_Variant_init(&xxx);
    UA_StatusCode us=UA_Server_readValue(server,*nodeid,&xxx);
    printf("%i\n",us);
}

closed 2025-02-04 by teccheck