open62541/open62541 #3305: [question] subscription/monitoreditems performance。

jixian79, 2019-11-29:
## Description
I have created a OPCUA Server with many variables to monitor, there are 9 Object and each object contains about 63 variable.In Client , for each variable i create one monitoreditems with sampling rate at 250ms. In the test environment , i only use one Object, 63 variablesto monitor。But when i created the monitoreditems, the opcua server respond very slowly。for example, after created these monitoreditems, when i use UaExpert to view these variable, it uses about 1 minute to connect to the opcua server i created。If I don't created the monitoreditems, UaExpert browse the variables smoothly。

Each variable I use a datasource to read the data from PLC through modbus or snap7 protocol。The test environment run on the loopback network , modbus and snap7 are the same。

## Background Information / Reproduction Steps

the Server run like this:
```
   while(running) {
       UA_UInt16 timeout = UA_Server_run_iterate(server, waitInternal);
       QThread::msleep(timeout);
   }
   retval = UA_Server_run_shutdown(server);
```
and i have tried to decrease the timeout, no obvious improvement 。

I run the client like this:
```
qDebug()<<"Worker thread is "<<QThread::currentThreadId()<<endl;
UA_CreateSubscriptionRequest request = UA_CreateSubscriptionRequest_default();
request.requestedPublishingInterval = 200;
request.requestedMaxKeepAliveCount = 20;
UA_CreateSubscriptionResponse response = UA_Client_Subscriptions_create(_client, request,&_machine_code, NULL,deleteSubscriptionCallback);

if(response.responseHeader.serviceResult != UA_STATUSCODE_GOOD)
{
	UA_Client_disconnect(_client);
	UA_Client_delete(_client);
	qWarning()<<"Create Subscribe failed\n";
	return ;
}

UA_UInt32 subId = response.subscriptionId;
qInfo()<<"Create subscription succeeded, id "<<subId<<"\n";

QByteArray name = _machine_code.toLatin1();
listTreeRecursive(_client, UA_NODEID_STRING(1, name.data()), &response, handler_monitored);

batchCreateMonitoredItems(_client, subId, handler_monitored, _db))

UA_StatusCode retval;
while(!QThread::currentThread()->isInterruptionRequested())
{
	retval = UA_Client_connect_username(_client, _endpointUrl.toLatin1(), _username.toLatin1().data(), _password.toLatin1().data());
	if (retval != UA_STATUSCODE_GOOD)
	{
		UA_LOG_ERROR(UA_Log_myout, UA_LOGCATEGORY_USERLAND,
			"Not connected. Retrying to connect in 1 second");
		/* The connect may timeout after 1 second (see above) or it may fail immediately on network errors */
		/* E.g. name resolution errors or unreachable network. Thus there should be a small sleep here */
		UA_sleep_ms(1000);
		continue;
	}
	retval = UA_Client_run_iterate(_client, 50);		
	if (retval != UA_STATUSCODE_GOOD)
	{
		qWarning() << "Client stopped unexpected, errorcode:" << retval << "\n";
		break;
	}
}
```
another question is , when the client run , a lot of 
```
[2019-11-29 15:53:08.413 (UTC+0800)] error/client       Inactivity for Subscription 1.
```
and i konw it's about subscription maxlifetime and maxkeeplivecount, but how can i 
solve this error? i use the default subscription parameters。I have read some performance test
result of open62541, so i think maybe i do something wrong cause the slowly response time. When i change the value, client monitored variable has to  take about 2~3 second to response. 

## Checklist

Please provide the following information:

 -  open62541 Version (release number or git tag): 1.0.0
 -  Other OPC UA SDKs used (client or server): UaExpret
 -  Operating system: Windows
 -  Logs (with `UA_LOGLEVEL` set as low as necessary) attached [log20191129.log](https://github.com/open62541/open62541/files/3904318/log20191129.log)
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

schroeder-, 2019-11-29:
Did you measure the time 1 IO needs on your server? Because open62541 is single threaded and the server will try all 250ms to read all variables in subscription before it continues to respond to any client. So the server will block until all 63 variables are read.

jixian79, 2019-11-29:
@schroeder-  ok, i will measure it 。And  is the multithread experiment feature can help this ?  what can i do if i want  monitored the data change in realtime about 100ms？

ccvca, 2019-12-02:
> Each variable I use a datasource to read the data from PLC through modbus or snap7 protocol。The test environment run on the loopback network , modbus and snap7 are the same。

I suspect that you are reading synchronously on the PLC, which take some time per variable (e.g. 20ms). So reading 63 will probably exceed the 250ms.

closed 2020-04-15 by jixian79