open62541/open62541 #3313: Setting up a server connection with securityPolicyUri "http://opcfoundation.org/UA/SecurityPolicy#Basic256"

bavilla, 2019-12-04:
Hi!
I use version 0.3 with UA_ENABLE_ENCRYPTION. 
I create a client with default parameters:
UA_Client *client = UA_Client_new(UA_ClientConfig_default);

Connect to the server to get UA_Client_getEndpoints() and get the UA_EndpointDescription I need. Its contents:

securityMode UA_MESSAGESECURITYMODE_NONE
securityPolicyUri "http://opcfoundation.org/UA/SecurityPolicy#None"

userIdentityTokensSize 4

userIdentityTokens[0].policyId "0"
userIdentityTokens[0].tokenType UA_USERTOKENTYPE_ANONYMOUS
userIdentityTokens[0].securityPolicyUri "http://opcfoundation.org/UA/SecurityPolicy#Basic256"

userIdentityTokens[1].policyId "1"
userIdentityTokens[1].tokenType UA_USERTOKENTYPE_USERNAME
userIdentityTokens[1].securityPolicyUri "http://opcfoundation.org/UA/SecurityPolicy#Basic256"

userIdentityTokens[2].policyId "2"
userIdentityTokens[2].tokenType UA_USERTOKENTYPE_CERTIFICATE
userIdentityTokens[2].securityPolicyUri "http://opcfoundation.org/UA/SecurityPolicy#Basic256"

userIdentityTokens[3].policyId "3"
userIdentityTokens[3].tokenType UA_USERTOKENTYPE_ISSUEDTOKEN
userIdentityTokens[3].issuedTokenType "urn:oasis:names:tc:SAML:1.0:assertion:Assertion"
userIdentityTokens[3].securityPolicyUri "http://opcfoundation.org/UA/SecurityPolicy#Basic256"

The next time I call UA_Client_connect(), I get the error "No suitable UserTokenPolicy found for the possible endpoints"

Tell us how to connect correctly to the server that has this configuration. I suppose that you need to configure UA_ClientConfig when initializing the client, but I did not understand from the examples and help how to do it.

jpfr, 2019-12-06:
Do you load the client certificate and private key?

See here: https://github.com/open62541/open62541/blob/master/examples/encryption/client_encryption.c

The client chooses a possible endpoint automatically.
You can filter the endpoints by these settings: https://github.com/open62541/open62541/blob/34a46e44e4e046f7e590d0be5745898f5290099d/include/open62541/client_config.h#L64

bavilla, 2019-12-16:
Thank you for the comment. I haven't solved this issue yet. I use VC2019. I must have built mbedtls incorrectly, catching exceptions when using this library. I will study it further

closed 2021-03-20 by bavilla