open62541/open62541 #3318: Proposal: Disable readspeed tests by default

mlgiraud, 2019-12-09:
## Description
On the master branch we have a test check_server_readspeed.
I think we had this as an example before. Why was this moved to unit tests?
I don't see the value we get by running this on each CI run. It takes a lot of time and does not give us any valuable results, except if inspected manually (is the speed still ok? although this also could vary depending on architecture changes of the ci runners?).
I would propose moving this back to the examples, or maybe disabling it for CI to quicken the test runs.
When testing locally this also consumes a lot of time, without giving any valuable information in most cases.
We could add a cmake option that enables the test when desired.
Then we can run local tests with benchmarking enabled if desired.

jpfr, 2019-12-09:
The speed tests use internal API.
Hence tests and not examples.

It would be good if we could track speed regressions semi-automatically.

As a workaround we can set the number of repetitions to something small if debugging is enabled.

mlgiraud, 2019-12-11:
Maybe we could create a ci run that only runs speed tests and is marked as "may fail", and remove the speed tests from all the other ci runs. That way we could maybe include an assertion that the time taken has to be similar to the last run.
Also, don't you think it is enough to run the speed tests on one ci run, and not all of them?

teccheck, 2025-02-04:
This seems to be outdated. Please reopen if needed.

closed 2025-02-04 by teccheck