open62541/open62541 #3319: Engineering Units to BASEDATAVARIABLETYPE

Maerdl, 2019-12-10:
Hi there,

i have a question concerning the engineering units.
Let's say i have a data variable created with UA_Server_addDataSourceVariableNode(); its type definition is `UA_NS0ID_BASEDATAVARIABLETYPE.`
So far so good. After that i have attached the Engineering Unit as property by calling something like that:
`UA_Server_addVariableNode(server, UA_NODEID_NUMERIC(1, x++), _analogItemNodeReturned, UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY), UA_QUALIFIEDNAME(0, "EURange"), UA_NODEID_NUMERIC(0, UA_NS0ID_PROPERTYTYPE), attr, NULL, &_enUnitNodeRange );`

I can see the engineering unit property but and the datatype as well is fine, but i can not read the engineering unit variable, it returns a NULL.

Any ideas to get this done ? Or is it not allowed to attach engineering unit property to a base data variable type ?

Thanks in advance
Maerdl

jpfr, 2019-12-11:
Did you set a value for the EngineeringUnit in the `attr` field?

Maerdl, 2019-12-12:
I tried the following:

created a EU Information:
` UA_EUInformation eu;
 eu.namespaceUri = UA_STRING("somecompany.com");
 eu.description = UA_LOCALIZEDTEXT("en-US", "meter per second");
 eu.displayName = UA_LOCALIZEDTEXT("en-US", "m/s");
 eu.unitId = -1; `

initialize the attribtes:
 `UA_VariableAttributes attr; 
 UA_VariableAttributes_init(&attr);`

create the node which will be created, and add attributes
` UA_NodeId _enUnitNodeUnits;
 attr.dataType = UA_TYPES[UA_TYPES_EUINFORMATION].typeId;
 attr.userWriteMask = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
 UA_Variant_setScalarCopy(&attr.value, &eu, &UA_TYPES[UA_TYPES_EUINFORMATION]); `

Create the Has Property from the base node where _analogItemNodeReturned is the parent node of the property:
` status = UA_Server_addVariableNode(server, UA_NODEID_NUMERIC(1, x++), _analogItemNodeReturned, UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY), UA_QUALIFIEDNAME(0, "EngineeringUnits"), UA_NODEID_NUMERIC(0, UA_NS0ID_PROPERTYTYPE), attr, NULL, &_enUnitNodeUnits );`

I can see the EngineeringUnits Property attached to the node `_analogItemNodeReturned`.
If i click the property i can not read it's value.
[image]

Do i have to add a variable node which contains the information ?

Thanks in advance!

jpfr, 2019-12-18:
Try to initialize the variable attributes as

```c
UA_VariableAttributes attr = UA_VariableAttributes_default;
```

This sets some sane defaults. For example read permissions.

Maerdl, 2019-12-20:
Thank's a lot. That works.
Thought UA_VariableAttributes_init(&attr) does the default settings. After looking into it is clear that this is not done...

Again, thanks

closed 2019-12-20 by Maerdl