open62541/open62541 #3320: why UA_PubSubTransportLayer() did not define？

Dingli16, 2019-12-11:
## Description

	config->pubsubTransportLayers[0] = UA_PubSubTransportLayerUDPMP();
		config->pubsubTransportLayersSize++;
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-12-11:
Can you explain a bit more?

teccheck, 2025-02-04:
Seems to be abandoned. Feel free to reopen if needed.

closed 2025-02-04 by teccheck