open62541/open62541 #3322: Couldn't get all child nodes when browsing

Zvicii, 2019-12-16:
I try to browse a siemens server.
I managed to get noded under UA_NS0ID_ROOTFOLDER using UA_Client_Service_browse.
But there is a folder that I couldn't get its children.
the referenceSize is 0;

But UA_EXPERT can get children under it.
and UA_EXPERT can't get the folder's attributes, they are all red and the values are all'801f0000'

<sub>Sent from <a href="https://apps.apple.com/cn/app/id1314212521">PPHub</a></sub>

Zvicii, 2019-12-16:
[image]

RantiE, 2019-12-17:
UA_STATUSCODE_BADUSERACCESSDENIED 0x801F0000
/* User does not have permission to perform the requested operation. */

Have you got permission to browse the server?

Zvicii, 2019-12-17:
> UA_STATUSCODE_BADUSERACCESSDENIED 0x801F0000
> /* User does not have permission to perform the requested operation. */
> 
> Have you got permission to browse the server?

im not sure，but if UA_EXPERT can browse the nodes.under it, why cant I?
the settings are the same.

Zvicii, 2019-12-17:
also when i user ua_client_service_browse，the response header has status UA_STATUS_GOOD,if i dont have permission, the header should be different right? but stiil it returns with no nodes

jpfr, 2019-12-17:
How do you browse?
Do you constrain the reference types used for browsing?

Also, reading attributes is done with a different service.
`Read`, not `Browse`.

Zvicii, 2019-12-18:
I haven't set any constraint，other nodes like "Types"、"Views"、"Server" work just fine. But this node I want to browse doesn't return any child, the response_header means at least I have permission to browse right? even if I dont have permission to read;

Zvicii, 2019-12-18:
[image]

Zvicii, 2019-12-18:
[image]

jpfr, 2019-12-18:
The BrowseDescription is incomplete. You should set `referenceTypeId` and possibly `includeSubtypes`.

https://open62541.org/doc/current/types_generated.html#browsedescription

```c
typedef struct {
    UA_NodeId nodeId;
    UA_BrowseDirection browseDirection;
    UA_NodeId referenceTypeId;
    UA_Boolean includeSubtypes;
    UA_UInt32 nodeClassMask;
    UA_UInt32 resultMask;
} UA_BrowseDescription;
```

Zvicii, 2019-12-18:
> The BrowseDescription is incomplete. You should set `referenceTypeId` and possibly `includeSubtypes`.
>
> https://open62541.org/doc/current/types_generated.html#browsedescription
>
> ```c
> typedef struct [...]

thank you for reply so quickly；
I'll try and get back to you.

Zvicii, 2019-12-18:
> The BrowseDescription is incomplete. You should set `referenceTypeId` and possibly `includeSubtypes`.
>
> https://open62541.org/doc/current/types_generated.html#browsedescription
>
> ```c
> typedef struct [...]

what referenceTypeId should I set to get everything？
UA_NS0ID_REFERENCES?

Zvicii, 2019-12-18:
I set [includeSubtypes] to true and [refrenceTypeId] to UA_NODEID_NUMERIC(0, UA_NS0ID_REFERENCES). Still not working;
I appears that I can't get nodes which are obejcttype added manually;

Zvicii, 2019-12-18:
[image]
[image]
[image]
[image]

DeviceType1 DeviceType2 DeviceType3
I cant get these 3 nodes

Zvicii, 2019-12-18:
[image]

still not working

jpfr, 2019-12-19:
Difficult to say from here.

We have a unit test for the client browsing.
https://github.com/open62541/open62541/blob/master/tests/client/check_client_highlevel.c#L251

It is definitely working threre.
The status code UA_STATUSCODE_BADUSERACCESSDENIED hints at another problem.
Did you see my response from #3319?

> Try to initialize the variable attributes as

> UA_VariableAttributes attr = UA_VariableAttributes_default;

> This sets some sane defaults. For example read permissions.

Zvicii, 2019-12-20:
I made a mistake;

> The BrowseDescription is incomplete. You should set `referenceTypeId` and possibly `includeSubtypes`.
>
> https://open62541.org/doc/current/types_generated.html#browsedescription
>
> ```c
> typedef struct [...]

I made a mistake.
this works!
you guys are the best, love you!

closed 2019-12-20 by Zvicii