open62541/open62541 #3323: Question: Repository buildsystem/release strategy

MartinaFelberFEM, 2019-12-16:
Hello!

I would like to ask how your repository release-branch strategy looks like.

I am implementing an OpcUa server with PubSub and need to release a stable version.
I have mirrored your repository and created a branch on v1.0 tag with my own build system extensions to build your library. My server implementation depends on the v1.0 tagged library at the moment. 

Thats fine, but I am interested in new PubSub features too. So I thought that I could create another branch based on your master branch for my build system.

But I have noticed that some of your build files from master branch are different from branch 1.0 or tag v1.0.
E.g. the debian/control-template file configures different debian packages in master and v1.0 tag.
It seems that the buildsystem of the v1.0 tag version is more comprehensive than the current master branch.
On the other hand the 1.0 branch is 17 commits ahead, 314 commits behind master.

So now I don't know on which branch I should setup my build system for the upcoming release (1.0 || master), because I would like to use the latest PubSub features (fast-path, etc.) but I have to depend on a stable open62541 version.
 
I am not sure what's your "git flow"?
Are all commits of branch 1.0 in the next release 1.1 for sure?
And the master branch has all features, but maybe not all of them are available in the next release? So I can't depend on the master branch, because some features may be not available in release 1.1?

Many thanks in advance for your support.

Best regards
Martina

jpfr, 2019-12-16:
> Are all commits of branch 1.0 in the next release 1.1 for sure?

Yes. We apply important fixes to 1.0 and then do a `git merge 1.0` from master.

The 1.1 branch will be forked from master.

https://github.com/open62541/open62541/wiki/Creating-a-new-release

jpfr, 2019-12-16:
> But I have to depend on a stable open62541 version

In that case, I would wait until we branch off 1.1 and freeze the API.

I guess that will be in early January.
But the core devs will decide on this together.

MartinaFelberFEM, 2019-12-17:
Many thanks for your information! :)

closed 2019-12-17 by MartinaFelberFEM