open62541/open62541 #3328: Alarm & Conditions: Messages are changed when EnableState is set

ccvca, 2019-12-19:
## Description

When changing the EnableState Member (by code), the Message of the Condition is replaced. It was quite unexpected for me that my previous message was not available and I could not find anything about this in the specification.

Looking at the code, this is also done when adding a comment, which looks weird as client will never be able to correlate the original message witht the comment if it has not been saved before (e.g. connected after comment has been added).

## Background Information / Reproduction Steps

https://github.com/open62541/open62541/blob/ff87c2876d8e53b7f09472a90c731daa789e9a28/src/server/ua_subscription_alarms_conditions.c#L77

https://github.com/open62541/open62541/blob/ff87c2876d8e53b7f09472a90c731daa789e9a28/src/server/ua_subscription_alarms_conditions.c#L760-L764

Used CMake options:

BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_GNUtoMS:BOOL=OFF
CMAKE_INSTALL_PREFIX:PATH=D:/Programmieren/Cpp/Open62541_Tests/install-debug
UA_ARCHITECTURE:STRING=win32
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_ENCRYPTION:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_ENABLE_WEBSOCKET_SERVER:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=FULL

## Checklist
 - [x] open62541 Version (release number or git tag): master / ff87c2876d8e53b7f09472a90c731daa789e9a28
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Win 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

sqadasi, 2020-02-04:
The message describes the event which was triggered. So when enabling a condition or adding a comment, the server triggers an event with a description about it. 
You are right that it messages about enabling, comments, etc. maybe unnecessary information which will overwrite more important messages, but still, just not updating the message on such occasions won't solve the problem when messages get overwritten by other events from the application itself.
In my opinion, historizing all condition events would be a better solution.

ccvca, 2020-02-04:
>  In my opinion, historizing all condition events would be a better solution.

But in case a Client connects to an OPC UA-server with an active Condition, that has been set a comment, this client can only get the Message "Comment set", instead of the original one, that was overrideen before the client connects to the server.

Updating a message from an application is always fine, as the application has full control about this and can decide for/ against overriding this information. But as overriding is inside the OPC UA-SDK and can be triggered by a client, a server-application has no chance to preserve the original message for future clients.

sqadasi, 2020-05-20:
@ccvca you are right. Actually, the current implementation of A&C has many limitations for the application, and does a lot of "unnecessary (like setting message field)" work under the hood. The original idea was to implement A&C in such a way, that the application has minimal amount of work to do, but such thing will always fire back because it takes the flexibility away from the application. Not to mention the confusion it causes.
A major rework for the A&C module is inevitable in my opinion.
Functionalities like:
- triggering alarms automatically when writing in specific Condition Fields or Field Properties.
- writing default messages.
- creating HasCondition and HasEventNotifier/Source references automatically.
- etc
should be removed, so the application will do the work by itself.

ccvca, 2020-06-04:
@sqadasi One more question:
Why do all UA_Server_##Condition-Methods require the conditionSource as an argument? This might be read from the SourceNode instead?

https://github.com/open62541/open62541/blob/2087b670cb1dfe437b004c7ff1dc4bf5e6d74091/src/server/ua_subscription_alarms_conditions.c#L1923-L1932

sqadasi, 2020-06-04:
@ccvca if I haven't forgotten any function, the conditionSource argument is passed to the following public API functions:
1) UA_Server_createCondition
2) UA_Server_triggerConditionEvent
3) UA_Server_setConditionTwoStateVariableCallback
The idea here was to otpimize those functioncalls because reading the SourceNode Condition Field takes a bit longer. 
However, reading the SourceNode Condition Field is used in some internal calls (mainly in Variable after write callbacks). But as I stated in my last comment, those callbacks are putting a huge limitation on the A&C module usability and should be therefore removed or at least optimized.

hope that answers your question:)