open62541/open62541 #3329: Too high memory usage (above 60kB)

MM-Minh, 2019-12-19:
- [x] open62541 Version (release number or git tag): v1.0
 - [x] Operating system: FreeRTOS

I configured a hook for malloc errors in FreeRTOS and maximized the heap to something below 60kB. I used the Heap 4 implementation of FreeRTOS. Also, I tried to run the open62541 Server with minimal namespace (set in the cmake build). Well, I am not sure if I followed every step of the manual.
The program could be installed with less than 200kB space (which positively surprised me).
With that amount of heap FreeRTOS starts and then in the initialization of a minimal namespace 0, the OPC UA server will call the malloc error handler.

Can I do something else to lower the memory usage?

Have a nice day, 
Minh

Pro, 2019-12-19:
Did you already see this:
https://open62541.org/doc/current/building.html#minimizing-the-binary-size

An additional place where you could reduce heap usage is the size of send and receive Buffer Size.
(2 times 65kB).
See also:
https://github.com/Pro/open62541-esp32/blob/ca014bb51409b0280a0747698cd09ca355f5a066/main/main.c#L84

MM-Minh, 2019-12-20:
Thank you for responding.

> https://open62541.org/doc/current/building.html#minimizing-the-binary-size

I did forget to set this compile setting as seen in the following screenshot. 

[image]

This hint may make a change. But for development reasons such as debug symbol missing or less logs, this needs to be reevaluated. Also, in the embedded software the symbols can be stripped with strip from binutils or gcc, I believe I read somewhere.

> https://github.com/Pro/open62541-esp32/blob/ca014bb51409b0280a0747698cd09ca355f5a066/main/main.c#L84

Your post shows the following

```
    UA_UInt32 sendBufferSize = 16000;       //64 KB was too much for my platform
    UA_UInt32 recvBufferSize = 16000;       //64 KB was too much for my platform

    ESP_LOGI(TAG_OPC, "Initializing OPC UA. Free Heap: %d bytes", xPortGetFreeHeapSize());

    UA_Server *server = UA_Server_new();

    UA_ServerConfig *config = UA_Server_getConfig(server);

    UA_ServerConfig_setMinimalCustomBuffer(config, 4840, 0, sendBufferSize, recvBufferSize);
```

And I did resize the send and receive Buffer Size to each 8kB.

```
> #define recv_bufferSize 8192
> #define send_buffersize recv_bufferSize
```

Later called in 
`UA_StatusCode return_value = UA_ServerConfig_setMinimalCustomBuffer(server_config, portnumber, 0, send_buffersize, recv_bufferSize);`

Great to see, we were on the same path.

Thank you very much again, but I think I will close my attempts until next year, if I have time. Merry Christmas and a happy new year.