open62541/open62541 #3330: Assert failed when triggering (too) many events or clients receive events too slow

ccvca, 2019-12-20:
## Description

When there are too many events in the queue to the client this assertation failed:
https://github.com/open62541/open62541/blob/63771722737a723f2f4352bdf03ae0c640a8395d/src/server/ua_subscription_monitoreditem.c#L275

Example numbers from my project:
mon->eventOverflows: 4
mon->queueSize: 6
=> 4 _(mon->eventOverflows)_ <= 6 _(mon->queueSize)_ - 4 _(mon->eventOverflows)_ + 1
=> 4 <= 3 (FAIL)

I triggered this consistently in my application, by producing 2 events every second after connecting by UaExpert and subscribe the Server object. This freezes UaExpert for several seconds (looks like creating the subscriptions in a blocking manner, so no publish-Messages are send) and then my application crashes.

## Background Information / Reproduction Steps
I modified the [tutorial_server_events](https://github.com/open62541/open62541/files/3987150/tutorial_server_events.zip) to create 6 (5 works fine) events every 2 seconds. Just connection by UaExpert crashes the application. with this assertation.

<details>
<summary>Embedded source code of modified event tutorial to trigger this assert, Click to reveal</summary>
    
    /* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
     * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

    #include <open62541/plugin/log_stdout.h>
    #include <open62541/server.h>
    #include <open62541/server_config_default.h>

    #include <signal.h>
    #include <stdlib.h>

    /**
     * Generating events
     * -----------------
     * To make sense of the many things going on in a server, monitoring items can be useful. Though in many cases, data
     * change does not convey enough information to be the optimal solution. Events can be generated at any time,
     * hold a lot of information and can be filtered so the client only receives the specific attributes he is interested in.
     *
     * Emitting events by calling methods
     * ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
     * The following example will be based on the server method tutorial. We will be
     * creating a method node which generates an event from the server node.
     *
     * The event we want to generate should be very simple. Since the `BaseEventType` is abstract,
     * we will have to create our own event type. `EventTypes` are saved internally as `ObjectTypes`,
     * so add the type as you would a new `ObjectType`. */

    static UA_NodeId eventType;

    static UA_StatusCode
    addNewEventType(UA_Server *server) {
    	UA_ObjectTypeAttributes attr = UA_ObjectTypeAttributes_default;
    	attr.displayName = UA_LOCALIZEDTEXT("en-US", "SimpleEventType");
    	attr.description = UA_LOCALIZEDTEXT("en-US", "The simple event type we created");
    	return UA_Server_addObjectTypeNode(server, UA_NODEID_NULL,
    									   UA_NODEID_NUMERIC(0, UA_NS0ID_BASEEVENTTYPE),
    									   UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
    									   UA_QUALIFIEDNAME(0, "SimpleEventType"),
    									   attr, NULL, &eventType);
    }

    /**
     * Setting up an event
     * ^^^^^^^^^^^^^^^^^^^
     * In order to set up the event, we can first use ``UA_Server_createEvent`` to give us a node representation of the event.
     * All we need for this is our `EventType`. Once we have our event node, which is saved internally as an `ObjectNode`,
     * we can define the attributes the event has the same way we would define the attributes of an object node. It is not
     * necessary to define the attributes `EventId`, `ReceiveTime`, `SourceNode` or `EventType` since these are set
     * automatically by the server. In this example, we will be setting the fields 'Message' and 'Severity' in addition
     * to `Time` which is needed to make the example UaExpert compliant.
     */
    static UA_StatusCode
[... 185 lines cut ...]
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_ENCRYPTION:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_ENABLE_WEBSOCKET_SERVER:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=FULL

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master / 63771722737a723f2f4352bdf03ae0c640a8395d
 - [x] Other OPC UA SDKs used (client or server): UaExpert 1.5.0
 - [x] Operating system: Win 10
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached (added in issue)
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue: Might break Servers, when (bad) clients do not send enought publish-messages to read events fast enought

Pro, 2019-12-20:
We (myself and @aDogCalledSpot) also ran into the same issue a few days ago.
@aDogCalledSpot is already working on a fix, it will land soon.

The issue only arises, if DiscardOldest is set to false and the client doesn't query the events fast enough. In that case, the events pile up in the queue and as soon as the queue is full, the server is supposed to add overflow events.

If DiscardOldest is true, then the OverflowEvent is just added at the beginning of the queue and new overflow events are discarded, as long as there is already an overflow event at that place.

If DiscardOldest is false, the overflow event is added at the end of the queue. And here is currently a bug which adds multiple OverflowEvents after each other without an event in between. That is causing the assert.

jpfr, 2019-12-22:
Very cool @pro

PR #3331 (2019-12-24): Limit overflow events
Fixes #3330
closed 2019-12-24 by Pro