open62541/open62541 #3337: Does PubSub support custom data type?

chuongvhn, 2019-12-30:
## Description
In Publisher's side I'm able to make my custom data type available to server then publish value of the custom data type. 

In Subscriber size I always receive 0.

I saw that `UA_DataSetMessage_decodeBinary` cannot decode message because of `customTypes` is NULL. Tracing into the stack call I didn't see anywhere 'customTypes` can be passed. 

 

 

 

Looking to `ua_client.c` it does have some functions call with `customTypes` passed in, but I need to do kind of `pubsub connection -> addDataSetReader-> ...>subscribedTargetVariable` so the ua_client  cannot help.

 
[image]

Is there a way to subscribe to a published custom data type ? 
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

chuongvhn, 2020-01-02:
@jpfr so does it mean (by labeling as Enhancement) currently PuSub hasn't support subscription of custom data type yet ?

jpfr, 2020-01-07:
It doesn't have it atm.

But this should be easy to do. The information about custom datatypes is in the server.
We just need to put the additional pointer into the WriterGroup/ReaderGroup.

This feature is up for grabs by interested developers.

chuongvhn, 2020-01-08:
@jpfr : Could you pleas help to add more detail ?
As far as I know, currently in the subscriber side what's happening is subscriber - once receive packet - will `UA_NetworkMessage_decodeBinary` -> `UA_DataSetMessage_decodeBinary` ( see attached screen shot above ).   
And In `UA_DataSetMessage_decodeBinary` it will call `Variant_decodeBinary` then it detects that the payload data type is not build in type so it starts to look for type of customDataType.  To make customDataType to be available we need somewhere to pass in customDataType.    
In the setting up for WirterGroup an ReaderGroup + DataSetReader I don't know how should custom data tye is made available ?

This is the source of Publisher WirterGroup:  
```
    UA_NodeId connection_id_, 
              published_dataSet_id_, 
              dataSet_field_id_,
              writer_group_id_, 
              dataSet_writer_id_;

    UA_ReaderGroupConfig reader_group_cfg_;
    UA_DataSetReaderConfig dataset_reader_cfg_;
    UA_NodeId dataset_reader_id_, reader_group_id_;

    /* create a new WriterGroupConfig and add the group to the existing PubSubConnection. */
    UA_WriterGroupConfig writerGroupConfig;
    memset(&writerGroupConfig, 0, sizeof(UA_WriterGroupConfig));
    writerGroupConfig.name = UA_STRING("wrg1");
    writerGroupConfig.publishingInterval = 100; /* Publish interval in milliseconds */;
    writerGroupConfig.enabled = UA_FALSE;
    writerGroupConfig.writerGroupId = 100;
    writerGroupConfig.encodingMimeType = UA_PUBSUB_ENCODING_UADP;
    writerGroupConfig.messageSettings.encoding             = UA_EXTENSIONOBJECT_DECODED;
    writerGroupConfig.messageSettings.content.decoded.type = &UA_TYPES[UA_TYPES_UADPWRITERGROUPMESSAGEDATATYPE];

    /* The configuration flags for the messages are encapsulated inside the
     * message- and transport settings extension objects. These extension
     * objects are defined by the standard. e.g.
     * UadpWriterGroupMessageDataType */
    UA_UadpWriterGroupMessageDataType *writerGroupMessage  = UA_UadpWriterGroupMessageDataType_new();

    /* Change message settings of writerGroup to send PublisherId,
     * WriterGroupId in GroupHeader and DataSetWriterId in PayloadHeader
     * of NetworkMessage */
    writerGroupMessage->networkMessageContentMask = (UA_UadpNetworkMessageContentMask)(UA_UADPNETWORKMESSAGECONTENTMASK_PUBLISHERID |
                                                    (UA_UadpNetworkMessageContentMask)UA_UADPNETWORKMESSAGECONTENTMASK_GROUPHEADER |
                                                    (UA_UadpNetworkMessageContentMask)UA_UADPNETWORKMESSAGECONTENTMASK_WRITERGROUPID |
                                                    (UA_UadpNetworkMessageContentMask)UA_UADPNETWORKMESSAGECONTENTMASK_PAYLOADHEADER);
    writerGroupConfig.messageSettings.content.decoded.data = writerGroupMessage;
    UA_Server_addWriterGroup(server, connection_id_, &writerGroupConfig, &writer_group_id_);

    //FIXME: update open62541 to make API available
    //UA_Server_setWriterGroupOperational(server, &writer_group_id_);

    UA_UadpWriterGroupMessageDataType_delete(writerGroupMessage);
```  

Subscriber ReaderGroup and DatasetReader:  
```
void addReaderGroup(UA_Server *server) 
{
    if(server == NULL) {
        return;
    }

    memset ( &reader_group_cfg_ , 0, sizeof(UA_ReaderGroupConfig) );
    reader_group_cfg_.name = UA_STRING("rdg1");
    UA_Server_addReaderGroup(server, connection_id_, &reader_group_cfg_, 
                            &reader_group_id_);
}

void addDataSetReader(UA_Server *server) {
    if(server == NULL) {
        return;
    }

    memset (&dataset_reader_cfg_, 0, sizeof(UA_DataSetReaderConfig));
    dataset_reader_cfg_.name = UA_STRING("DataSet Reader 1");

    /* where subscription for a variable is specified */
    UA_UInt16 publisher_id = 2234;
    dataset_reader_cfg_.publisherId.type = &UA_TYPES[UA_TYPES_UINT16];
    dataset_reader_cfg_.publisherId.data = &publisher_id;
    dataset_reader_cfg_.writerGroupId = 100;
    dataset_reader_cfg_.dataSetWriterId = 62541;

    /* Setting up Meta data configuration in DataSetReader */
    UA_DataSetMetaDataType_init (&dataset_reader_cfg_.dataSetMetaData);
    pMetaData->name = UA_STRING ("dsr1");

    pMetaData->fieldsSize = 1;
    pMetaData->fields = (UA_FieldMetaData*)UA_Array_new (pMetaData->fieldsSize,
                         &UA_TYPES[UA_TYPES_FIELDMETADATA]);
    /*custom data Type*/
    UA_FieldMetaData_init (&pMetaData->fields[0]);
    UA_NodeId_copy(&UA_MY_CUSTOMTYPE[UA_MYCUSTOMTYPE_TYPE1].typeId,
                    &pMetaData->fields[0].dataType);
    pMetaData->fields[0].builtInType = UA_NS0ID_STRUCTURE;
    pMetaData->fields[0].name = UA_STRING("type1");
    pMetaData->fields[0].valueRank = UA_VALUERANK_SCALAR; /* scalar */
    
    UA_Server_addDataSetReader(server, reader_group_id_, &dataset_reader_cfg_,
                                          &dataset_reader_id_);
}

```

Thank you in advance !

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier