open62541/open62541 #3342: XML Nodeset Compiler not recognizing type <opc:Bit>

GrigoryFridman, 2020-01-07:
# Description

I am working on connecting a UWB RTLS system to OPC UA. For this occasion I am using the current AutoID Nodeset (v1.00) and its RTLSDeviceType, provided by the OPC Foundation. For translating the nodeset I am using the open62541 XML Nodeset Compiler. The problem is, that the bsd-file ([types.txt](https://github.com/open62541/open62541/files/4030339/types.txt)) defining the custom datatypes for AutoID is using the type "opc:Bit" (example:

```xml
</opc:StructuredType>
 <opc:StructuredType BaseType="ua:ExtensionObject" Name="ScanResult">
  <opc:Field TypeName="opc:Bit" Name="LocationSpecified"/>
  <opc:Field Length="31" TypeName="opc:Bit" Name="Reserved1"/>
  <opc:Field TypeName="opc:CharArray" Name="CodeType"/>
  <opc:Field TypeName="tns:ScanData" Name="ScanData"/>
  <opc:Field TypeName="opc:DateTime" Name="Timestamp"/>
  <opc:Field SwitchField="LocationSpecified" TypeName="tns:Location" Name="Location"/>
 </opc:StructuredType>
```

 ) for defining the optional fields, which the compiler does not recognize:

```bash
Traceback (most recent call last):
  File "/home/grigory/Abschlussprojekt/open62541-test/open62541/tools/generate_datatypes.py", line 83, in <module>
    parser.create_types()
  File "/home/grigory/Abschlussprojekt/open62541-test/open62541/tools/nodeset_compiler/type_parser.py", line 275, in create_types
    self.parse_types()
  File "/home/grigory/Abschlussprojekt/open62541-test/open62541/tools/nodeset_compiler/type_parser.py", line 306, in parse_types
    self.parseTypeDefinitions(self.outname, f, self.namespace)
  File "/home/grigory/Abschlussprojekt/open62541-test/open62541/tools/nodeset_compiler/type_parser.py", line 239, in parseTypeDefinitions
    "E.g. '--import==UA_TYPES#/path/to/deps/ua-nodeset/Schema/Opc.Ua.Types.bsd'")
RuntimeError: Infinite loop detected or type not found while processing types OcrScanResult: unknonwn subtype ['Bit', 'Bit', 'Bit', 'Bit'] Maybe you need to import additional types with the --import flag. E.g. '--import==UA_TYPES#/path/to/deps/ua-nodeset/Schema/Opc.Ua.Types.bsd'
```

I have written a workaround for my purpose not using this datatype, but I wanted to inform, that the XML Nodeset Compiler does not support such obsolete things like this datatype.

Used CMake options:

```bash
BUILD_SHARED_LIBS:BOOL=ON
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=ON
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_ENCRYPTION:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_ENABLE_WEBSOCKET_SERVER:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=FULL
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-01-07:
@andreasebner has looked into the same issue.
Can you say a word on the status?

GrigoryFridman, 2020-01-07:
Hey @jpfr, maybe I should have mentioned that I am the student working with/for @andreasebner on this problem.

o6-nbeier, 2026-10-20:
Thank you for taking the time to open this issue and describe it so clearly.

The problem has since been fixed on the maintained branches (see #3503), so we are closing it as resolved.

closed 2026-10-20 by o6-nbeier