open62541/open62541 #3344: Question PubSub: addRepeatedCallback

MartinaFelberFEM, 2020-01-07:
Hello!

I would like to ask a question about the `addRepeatedCallback` interface parameters.

I am developing a PubSub implementation and I would like to configure multiple publishers and subscribers at startup and at runtime.

I have overwritten the `addRepeatedCallback` function and handle the publishing and subscribing in custom realtime tasks. I have a task function for WriterGroups and a different task function for ReaderGroups.

The WriterGroup tasks have to update the publisher node values from realtime process variables before invoking the WriterGroup callback, which creates the network message.
The ReaderGroup tasks have to invoke the receive callback and write the received values to realtime process variables.

Here's my problem:
`addRepeatedCallback` is called when a Writer- **or** ReaderGroup is added to a connection. But the callback paramters do not provide any information if the current callback is a Writer- or ReaderGroup.

Therefore I can't know if I have to create a Writer- or ReaderGroup task. 

It seems that the parameter `UA_UInt64 *callbackId` is not used at the moment.
Is this parameter intended to differentiate between Writer- and ReaderGroups?
Or is there any other mechanism?

I would like to use the callback mechanism for subscribers, because it's more convenient than handling all received networkmessage myself as shown in following example: https://github.com/open62541/open62541/blob/4e38d391dde9e5d74463f0a17de245f8917675ca/examples/pubsub/tutorial_pubsub_subscribe.c

Many thanks for your support!
Best regards
Martina

jpfr, 2020-01-07:
The callbackid is used to delete callback later.

You could differentiate according to the callback method pointer.
This one is probably different (and known) for Writer- and ReaderGroup tasks.

MartinaFelberFEM, 2020-01-07:
Many thanks for your answer.
I have tried the following and it works:

```c
if (callback == (UA_ServerCallback) UA_ReaderGroup_subscribeCallback) {
   // create ReaderGroup task
} else if (callback == (UA_ServerCallback) UA_WriterGroup_publishCallback) {
   // create WriterGroup task
} else {
   // error
}
```

At first I thought it would be bad to check the address of the callback, but adding an additional parameter to the interface (e.g. an enum) would not make it any better. I am going to use this approach.
Many thanks! :)

closed 2020-01-07 by MartinaFelberFEM