open62541/open62541 #3345: Optional fields not really implemented in the XML Nodeset Compiler

GrigoryFridman, 2020-01-07:
## Description
I am working on connecting a UWB RTLS system to OPC UA. For this occasion I am using the current AutoID Nodeset (v1.00) and its RTLSDeviceType, provided by the OPC Foundation. For translating the nodeset I am using the open62541 XML Nodeset Compiler. The problem is, that the AutoID XML nodeset is using the Attribute `isOptional`, which the nodeset compiler does not recognize and open62541 does not support. This is an example for such a datatype, which uses this attribute:

```xml
<UADataType NodeId="ns=1;i=3010" BrowseName="1:ScanSettings">
        <DisplayName>ScanSettings</DisplayName>
        <References>
            <Reference ReferenceType="HasEncoding">ns=1;i=5015</Reference>
            <Reference ReferenceType="HasEncoding">ns=1;i=5016</Reference>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=22</Reference>
        </References>
        <Definition Name="1:ScanSettings">
            <Field DataType="Duration" Name="Duration"/>
            <Field DataType="Int32" Name="Cycles"/>
            <Field DataType="Boolean" Name="DataAvailable"/>
            <Field IsOptional="true" DataType="LocationTypeEnumeration" Name="LocationType"/>
        </Definition>
</UADataType>
```
Here it is the field `LocationType` of the datatype `ScanSettings`, that is optional. The main method for the AutoID devices is `Scan`, which gets a ScanSettings object as an input argument. If you call this method without using the LocationType in a client (e.g. UaExpert), the input argument I get in my code is 4 bytes shorter than it would be with it, so it gets decoded wrong by the generated function `UA_ScanSettings_decodeBinary()`. For my purposes I have written a workaround for this problem by making a new 4 byte longer `ByteString` and copying the too short one into that new one:

```c
UA_ExtensionObject *eo = (UA_ExtensionObject *) input[0].data;
    UA_ScanSettings settings;
    UA_ScanSettings_init(&settings);
    size_t scanSettingsOffset = 0;
    UA_ByteString byteString = eo->content.encoded.body;

    if(eo->encoding == UA_EXTENSIONOBJECT_ENCODED_BYTESTRING) {

        /* workaround for optional field issue if optional field not
         * included append empty bits to match expected decode size */
        if (eo->content.encoded.body.length <= 17) {
            UA_Byte *newBufferReceive = (UA_Byte *) UA_calloc(21, sizeof(UA_Byte));
            memcpy(newBufferReceive, eo->content.encoded.body.data, 17);
            eo->content.encoded.body.data = newBufferReceive;
            eo->content.encoded.body.length = 21;
            byteString = eo->content.encoded.body;
            byteString.length = 21;
        }
        UA_ScanSettings_decodeBinary(&byteString, &scanSettingsOffset, &settings);
    }
```

Summing up: because the nodeset compiler does not recognize the isOptional attribute and the bits with type <opc:Bit> used in the bsd-file for this optional fields (see #3342), although UaExpert does, arguments with optional fields can not get interpreted by open62541 right.

Used CMake options:

```bash
BUILD_SHARED_LIBS:BOOL=ON
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=ON
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_ENCRYPTION:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_ENABLE_WEBSOCKET_SERVER:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=FULL
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 18.04.3 LTS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-01-07:
Is this an issue to claim the subject or do you want feedback?

We should solve this the right way and not pile up workarounds.
I.e. we need proper support for structures with optional fields.

I think @andreasebner has some ideas.

jpfr, 2020-05-05:
Implemented and merged in #3503.

closed 2020-05-05 by jpfr