open62541/open62541 #3346: PubSub over MQTT: Build Error in Examples

waheedejaz, 2020-01-08:
## Description
Build error in pubsub examples with shared libs and mqtt.

```
Scanning dependencies of target pubsub_subscribe_standalone
[ 70%] Building C object examples/CMakeFiles/pubsub_subscribe_standalone.dir/pubsub/pubsub_subscribe_standalone.c.o
[ 72%] Linking C executable ../bin/examples/pubsub_subscribe_standalone
/tmp/ccKZEJMY.ltrans0.ltrans.o: In function `subscriberListen':
/home/waheed/uni/open62541/examples/pubsub/pubsub_subscribe_standalone.c:63: undefined reference to `UA_NetworkMessage_decodeBinary'
/home/waheed/uni/open62541/examples/pubsub/pubsub_subscribe_standalone.c:105: undefined reference to `UA_NetworkMessage_deleteMembers'
collect2: error: ld returned 1 exit status
examples/CMakeFiles/pubsub_subscribe_standalone.dir/build.make:95: recipe for target 'bin/examples/pubsub_subscribe_standalone' failed
make[2]: *** [bin/examples/pubsub_subscribe_standalone] Error 1
CMakeFiles/Makefile2:1260: recipe for target 'examples/CMakeFiles/pubsub_subscribe_standalone.dir/all' failed
make[1]: *** [examples/CMakeFiles/pubsub_subscribe_standalone.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

## Background Information / Reproduction Steps
Ubuntu 18.04 LTS

Used CMake options:
-DBUILD_SHARED_LIBS=TRUE
-DUA_BUILD_EXAMPLES=TRUE
-DUA_ENABLE_PUBSUB=TRUE
-DUA_ENABLE_PUBSUB_MQTT=TRUE

```bash
UA_ENABLE_PUBSUB_MQTT:BOOL=1
UA_BUILD_EXAMPLES:BOOL=1
UA_ENABLE_PUBSUB:BOOL=1
BUILD_SHARED_LIBS:BOOL=1
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ x] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ZekeLandon, 2020-01-08:
@waheedejaz Disable shared libraries and do clean build again [**-DBUILD_SHARED_LIBS=FALSE**]

waheedejaz, 2020-01-08:
I require dynamic linkage so I need the shared libs.

jpfr, 2020-01-08:
The linker is missing some methods.
These are internal and are not exported to the shared library.

If you need dynamic linkage, you can try to expose all methods via compiler flags or manually add the UA_EXPORT macro at the right places.

waheedejaz, 2020-01-08:
> These are internal and are not exported to the shared library.

Is it by design or is it bug? If I were to expose all the pubsub functions, would it be good idea to make a separate open62541_pubsub.so rather than to integrate it into main lib? Any thoughts?

closed 2020-01-10 by waheedejaz