open62541/open62541 #3350: PubSub Problem: Error to build PubSub with make command

BreemTUKL, 2020-01-09:
Hello,

I am new here and am currently completing my thesis on time sensitive networking with open62541.
I am currently having problems with the build PubSub using the make command (see screenshot).

can someone please help me?
Thanks in advance

Greetings Breem

Background information:
Operating system: Ubuntu 19.10 - 64-bit
- BUILD_SHARED_LIBS OFF
- UA_BUILD_EXAMPLES ON
- UA_ENABLE_AMALGAMATION OFF
- UA_NAMESPACE_ZERO FULL
- UA_ENABLE_PUBSUB ON
- UA_ENABLE_PUBSUB_ETH_UADP ON
Steps:
# enable additional features
sudo apt-get install cmake-curses-gui # for the ccmake graphical interface
sudo apt-get install libmbedtls-dev # for encryption support
sudo apt-get install check libsubunit-dev # for unit tests
sudo apt-get install python-sphinx graphviz # for documentation generation
sudo apt-get install python-sphinx-rtd-theme # documentation style

cd open62541
mkdir build
cd build
cmake ..
ccmake ..
make

## Checklist

Please provide the following information:

 - [ x] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

[open62541_cmake ...pdf](https://github.com/open62541/open62541/files/4040378/open62541_cmake.pdf)
[open62541_make.pdf](https://github.com/open62541/open62541/files/4040393/open62541_make.pdf)

ZekeLandon, 2020-01-10:
I followed your steps but I am **not able to reproduce** the error as shown in attachment.
Build is success for me. Can you confirm if you are using the [latest commit](https://github.com/open62541/open62541/commit/d02fc4f5a4b60bb30a3c9394643abdd93b1fd59c) from master branch?

BreemTUKL, 2020-01-10:
Hello ZekeLandon,

Thank you very much for your answer. I changed the src / server / ua_services_nodemanagement.c file as in the commit, but unfortunately the error still exists.
The error is in the file ua_pubsub_udp.c line 84 and line 104.
best regards
Breem

[image]
[image]

ZekeLandon, 2020-01-10:
@BreemTUKL, would you believe if I say this? I am stuck with different error in another machine.
Weirdo! By the way, what is the version of cmake and make used in your system?
Something related to compiler, I suspect. Post the results here and give a try in another machine.

**Version details in my PC:**
- CMake --> v3.5.1
- make   --> v4.1

[image]

Regards,
Zeke

BreemTUKL, 2020-01-11:
Hello @ZekeLandon,

Thank you for your reply.
So the problem was solved as follows, as you recommended to me.
I deleted the open62541 folder and created a new one, then made all the settings again and it worked :-)
But do you always have to delete the folders and install new ones if there is a problem? Is there no debug or anything to discover the errors?

thanks in advance for your help.
best regards
Breem

[image]

ccmake ..

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

ZekeLandon, 2020-01-12:
It's good that you're able to resolve the issue. I guess you need delete everything and clone again.
You could just delete your build folder and start again. That should be fine.

Cheers,
Zeke

BreemTUKL, 2020-01-14:
ok @ZekeLandon  thanks again for your help.

best regards
Breem

closed 2020-01-14 by BreemTUKL