open62541/open62541 #3351: Memory Leak ? : UA_Server_addNode_begin, UA_Server_addNode_finish

MartinaFelberFEM, 2020-01-10:
Hello!

I think there are memory leaks at following functions: 
UA_Server_addNode_begin
UA_Server_addNode_finish

I am using your current master branch (Commit SHA: d02fc4f5a4b60bb30a3c9394643abdd93b1fd59c) on Debian Jessie with following build options:

```bash
cmake .. -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_PUBSUB=YES -DUA_ENABLE_PUBSUB_ETH_UADP=YES -DUA_ENABLE_PUBSUB_CUSTOM_PUBLISH_HANDLING=YES -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=YES -DUA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS=YES
```

I wrote a PubSub application which can be configured, started and stopped multiple times at runtime. 
The server is always running. 

I always get 2 memory leaks when I add a node. 
I wasn't sure if it's my programming mistake, so I enabled the build option **UA_ENABLE_PUBSUB_INFORMATIONMODEL** which adds nodes when a new PubSub connection is setup and removes them when the connection gets removed.

It all works well and I can see that the configured PubSub connection nodes are visible under the Server object on load and get removed after an unload. The address space representation looks fine.
But there are leaks which all occur at methods **add**XXX**Representation** which are adding nodes to address space.

E.g.: https://github.com/open62541/open62541/blob/d02fc4f5a4b60bb30a3c9394643abdd93b1fd59c/src/pubsub/ua_pubsub_ns0.c#L227

addPubSubConnectionRepresentation
addPublishedDataItemsRepresentation
addDataSetWriterRepresentation
...

The leaks accumulate every time I load a new configuration.

```
aef59407   /home/admin/open62541_master/src/server/ua_nodes.c:540 - addReferenceKind
aef595c4   /home/admin/open62541_master/src/server/ua_nodes.c:581 - UA_Node_addReference
aeb5ae8a   /home/admin/open62541_master/src/server/ua_services_nodemanagement.c:1692 - addOneWayReference
aeb3adda   /home/admin/open62541_master/src/server/ua_server_utils.c:241 - UA_Server_editNode
aeb5b0e0   /home/admin/open62541_master/src/server/ua_services_nodemanagement.c:1743 - Operation_addReference
aeb57292   /home/admin/open62541_master/src/server/ua_services_nodemanagement.c:614 - addRef
aeb5817b   /home/admin/open62541_master/src/server/ua_services_nodemanagement.c:827 - AddNode_addRefs
aeb58ace   /home/admin/open62541_master/src/server/ua_services_nodemanagement.c:946 - Operation_addNode_begin
aeb5a154   /home/admin/open62541_master/src/server/ua_services_nodemanagement.c:1390 - UA_Server_addNode_begin
aeb4667a   /home/admin/open62541_master/src/pubsub/ua_pubsub_ns0.c:239 - addPubSubConnectionRepresentation
aeb445ec   /home/admin/open62541_master/src/pubsub/ua_pubsub_manager.c:91 - UA_Server_addPubSubConnection
```

```
aef5924e   /home/admin/open62541_master/src/server/ua_nodes.c:500 - addReferenceTarget
aef59656   /home/admin/open62541_master/src/server/ua_nodes.c:591 - UA_Node_addReference
aeb5ae8a   /home/admin/open62541_master/src/server/ua_services_nodemanagement.c:1692 - addOneWayReference
aeb3adda   /home/admin/open62541_master/src/server/ua_server_utils.c:241 - UA_Server_editNode
aeb5b0e0   /home/admin/open62541_master/src/server/ua_services_nodemanagement.c:1743 - Operation_addReference
aeb57292   /home/admin/open62541_master/src/server/ua_services_nodemanagement.c:614 - addRef
aeb582f9   /home/admin/open62541_master/src/server/ua_services_nodemanagement.c:840 - AddNode_addRefs
aeb56e68   /home/admin/open62541_master/src/server/ua_services_nodemanagement.c:529 - copyChild
aeb57042   /home/admin/open62541_master/src/server/ua_services_nodemanagement.c:570 - copyAllChildren
aeb57146   /home/admin/open62541_master/src/server/ua_services_nodemanagement.c:593 - addTypeChildren
aeb58f33   /home/admin/open62541_master/src/server/ua_services_nodemanagement.c:1003 - recursiveTypeCheckAddChildren
aeb59af5   /home/admin/open62541_master/src/server/ua_services_nodemanagement.c:1254 - AddNode_finish
aeb5a195   /home/admin/open62541_master/src/server/ua_services_nodemanagement.c:1399 - UA_Server_addNode_finish
aeb466eb   /home/admin/open62541_master/src/pubsub/ua_pubsub_ns0.c:250 - addPubSubConnectionRepresentation
aeb445ec   /home/admin/open62541_master/src/pubsub/ua_pubsub_manager.c:91 - UA_Server_addPubSubConnection
```

I don't know why the leaks occur, because every addXXXRepresentation method has a corresponding removeXXXRepresentation method, which looks fine. 
The method `UA_Server_deleteNode` seems to remove the node and the param `deleteReferences` is true.

I am using a custom memory management implementation (overwritten malloc, calloc, etc.) and a custom evaluation tool. 

I tested the same application with valgrind. And I got no memory leaks.
But I had to stop and exit the application to get an output from valgrind. 
So I think that the leaks get freed by the function `UA_Server_delete()`. 
Maybe at `UA_ServerConfig_clean` where the nodestore is cleaned up?
In my use case the server shall run "forever", so `UA_Server_delete` won't be called.

I have to dig deeper into the code to be sure about the leaks but I wanted to ask if someone knows about this.

Many thanks for your help!

More leaks:
[MemLeaks.txt](https://github.com/open62541/open62541/files/4044262/MemLeaks.txt)

MartinaFelberFEM, 2020-01-10:
I think that `addOneWayReference` is called twice as often as `deleteOneWayReference`.
Maybe that's the problem ... ?

I wrote a simple server test program.
On startup there are no custom defined nodes.
The program uses 2 additional signal handlers.
On signal SIGUSR1 I add 2 nodes and on signal SIGUSR2 I delete them.
The addressspace looks good.

I used linux perf tool to track the number of calls of deleteOneWayReference and addOneWayReference.

addOneWayReference gets called 16 times
deleteOneWayReference gets called 8 times

Test program:
[main.txt](https://github.com/open62541/open62541/files/4044775/main.txt)

Perf Output:
[perfData.txt](https://github.com/open62541/open62541/files/4044776/perfData.txt)

Steps to reproduce: (My executable name is `Publisher-open62541_nostrip`)
```
apt install linux-perf-4.9
perf probe -x ./Publisher-open62541_nostrip 'addOneWayReference'
perf probe -x ./Publisher-open62541_nostrip 'deleteOneWayReference'
perf probe -l
  probe_Publisher:addOneWayReference (on addOneWayReference@ua_services_nodemanagement.c in /home/admin/Publisher-open62541_nostrip)
  probe_Publisher:deleteOneWayReference (on deleteOneWayReference@ua_services_nodemanagement.c in /home/admin/Publisher-open62541_nostrip)
perf record -e 'probe_Publisher:addOneWayReference,probe_Publisher:deleteOneWayReference' -ag -p $(pidof Publisher-open62541_nostrip)

-----------------------------------------
send signals to program from another terminal:

kill -s USR1 $(pidof Publisher-open62541_nostrip)
kill -s USR2 $(pidof Publisher-open62541_nostrip)

-----------------------------------------
CTRL + C
perf script > /home/admin/perfData.txt
```

damijans, 2020-01-15:
Hi Martina,
I have checked your test program found in main.txt file. I am pretty sure the memory leak happens in usage of pumpID variable. In function addNodes() this nodeid is copied/cloned from the node which has been inserted into hash repository as a new node. 

Memory leak could be solved by explicitly freeing it in deleteNodes() function

```
static void deleteNodes(
   int sig) {
   serverMutexLock();
   UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "delete nodes");
   UA_Server_deleteNode(server, pumpId, UA_TRUE);
   UA_NodeId_clear(&pumpId);                                                                 // <=== free cloned pumpId
   UA_Server_deleteNode(server, manufacturerId, UA_TRUE);
   serverMutexUnlock();
}
```

or by not using it as an output nodeId (last argument) while calling UA_Server_addObjectNode() function from addNodes().

```
   UA_Server_addObjectNode(server, pumpId, ...., NULL, &pumpId);  

replace with 

   UA_Server_addObjectNode(server, pumpId, ...., NULL, NULL);  

```

MartinaFelberFEM, 2020-01-15:
Hi!
Many thanks for your help! :)

I removed the output nodeId at `UA_Server_addObjectNode` call as you suggested. 

```c
UA_Server_addObjectNode(server, pumpId, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
      UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT), UA_QUALIFIEDNAME(1, "Pump (Manual)"),
      UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE), oAttr, NULL, NULL);
```

Unfortunately, I still get the same result. 
I don't think that the nodeId is the problem.

I added some traces at add- and deleteOneWayReference method.
I traced the referenceItem and tried to match the output.

```
ooooooooooooooooooooooooooooooooooooooooooo- NO MATCH
addOneWayReference(): begin
item->sourceNodeId = 
	 Pump
item->referenceTypeId = 
	 47
item->isForward = 0
item->targetNodeId.nodeId = 
	 85
addOneWayReference(): end

47: UA_NS0ID_HASCOMPONENT
85: UA_NS0ID_OBJECTSFOLDER
Pump -> HasComponent -> ObjectsFolder
ooooooooooooooooooooooooooooooooooooooooooo- NO MATCH

xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx - MATCH
addOneWayReference(): begin
item->sourceNodeId = 
	 85
item->referenceTypeId = 
	 47
item->isForward = 1
item->targetNodeId.nodeId = 
	 Pump
addOneWayReference(): end

deleteOneWayReference(): begin
item->sourceNodeId = 
	 85
item->referenceTypeId = 
	 47
item->isForward = 1
item->targetNodeId.nodeId = 
	 Pump
deleteOneWayReference(): end
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx - MATCH

ooooooooooooooooooooooooooooooooooooooooooo- NO MATCH
addOneWayReference(): begin
item->sourceNodeId = 
	 Pump
item->referenceTypeId = 
	 40
item->isForward = 1
item->targetNodeId.nodeId = 
	 58
addOneWayReference(): end

40: UA_NS0ID_HASTYPEDEFINITION
58: UA_NS0ID_BASEOBJECTTYPE
Pump -> UA_NS0ID_HASTYPEDEFINITION -> UA_NS0ID_BASEOBJECTTYPE
ooooooooooooooooooooooooooooooooooooooooooo- NO MATCH

xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx - MATCH
addOneWayReference(): begin
item->sourceNodeId = 
	 58
item->referenceTypeId = 
	 40
item->isForward = 0
item->targetNodeId.nodeId = 
	 Pump
addOneWayReference(): end

deleteOneWayReference(): begin
item->sourceNodeId = 
	 58
item->referenceTypeId = 
	 40
item->isForward = 0
item->targetNodeId.nodeId = 
	 Pump
deleteOneWayReference(): end
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx - MATCH

ooooooooooooooooooooooooooooooooooooooooooo- NO MATCH
addOneWayReference(): begin
item->sourceNodeId = 
	 Pump King
item->referenceTypeId = 
	 47
item->isForward = 0
item->targetNodeId.nodeId = 
	 Pump
addOneWayReference(): end

Pump King -> UA_NS0ID_HASCOMPONENT -> Pump
ooooooooooooooooooooooooooooooooooooooooooo- NO MATCH

xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx - MATCH
addOneWayReference(): begin
item->sourceNodeId = 
	 Pump
item->referenceTypeId = 
	 47
item->isForward = 1
item->targetNodeId.nodeId = 
	 Pump King
addOneWayReference(): end

deleteOneWayReference(): begin
item->sourceNodeId = 
	 Pump
item->referenceTypeId = 
	 47
item->isForward = 1
item->targetNodeId.nodeId = 
	 Pump King
deleteOneWayReference(): end
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx - MATCH

ooooooooooooooooooooooooooooooooooooooooooo- NO MATCH
addOneWayReference(): begin
item->sourceNodeId = 
	 Pump King
item->referenceTypeId = 
	 40
item->isForward = 1
item->targetNodeId.nodeId = 
	 63
addOneWayReference(): end

Pump King -> UA_NS0ID_HASTYPEDEFINITION -> UA_NS0ID_BASEDATAVARIABLETYPE
ooooooooooooooooooooooooooooooooooooooooooo- NO MATCH

xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx - MATCH
addOneWayReference(): begin
item->sourceNodeId = 
	 63
item->referenceTypeId = 
	 40
item->isForward = 0
item->targetNodeId.nodeId = 
	 Pump King
addOneWayReference(): end

deleteOneWayReference(): begin
item->sourceNodeId = 
	 63
item->referenceTypeId = 
	 40
item->isForward = 0
item->targetNodeId.nodeId = 
	 Pump King
deleteOneWayReference(): end
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx - MATCH
```

It seems that the HAS_COMPONENT reference of a new node in backwards direction to the parent does not get deleted.
And the HAS_TYPEDEFINITION reference of a new node in forward direction to the type node.

But I don't know why. 
I have to add more traces.

Many thanks for your help!!

damijans, 2020-01-15:
> Unfortunately, I still get the same result.
> I don't think that the nodeId is the problem.
 
Probably we are talking about different issues. Namely, I didn't use your configuration settings and consequently probably I didn't reproduce your issue. However pumpId (how it is used in your description) represents memory issue. Technically speaking this is not memory leak since at the end apparent memory is freed. The problem is that while running addNodes(), deleteNodes() in a loop the memory consumption is constantly increasing. To reproduce this issue instead of sending USR signals (what is not reliable) use the following code:

```
int main(void)
{
   signal(SIGINT, stopHandler);
   signal(SIGTERM, stopHandler);
   server = UA_Server_new();
   UA_ServerConfig_setDefault(UA_Server_getConfig(server));
   while (running) {
          addNodes(1);
          deleteNodes(1);
   }
   UA_Server_delete(server);
}
```
... and monitor memory used by your application (for example watch  "sudo pmap `pidof <YOUR_APP>` | grep total")

In order to reproduce your issue probably I would have to use your configuration.

MartinaFelberFEM, 2020-01-16:
Many thanks for your advice and sorry for my sporadic answers.
I have monitored the memory consumption and it's not rising. Which is good, but I'm a bit confused now.
Maybe our memory evalutation tool has a bug. I am not sure.
I will investigate further.
Many thanks for your help!

juangburgos, 2020-01-21:
Could this be related to my issue #3387 ?

PR #3407 (2020-01-30): Shrink down memory buffer after removing target reference from UA_Node
Two independent issues were recently submitted complaining about memory leak after certain number of ua_nodes were removed from the server.
While analyzing this issue I have found out the problem is in removing target reference from node. Each node keeps list of references in two level list structure, where first list keeps track of different reference kind (for example references of parent node to its child node) while second list keeps track of target references of the same reference kind.
When new target reference is ins [... cut]
closed 2020-01-30 by jpfr