open62541/open62541 #3352: Can't build open62541 with UA_ENABLE_ENCRYPTION on win7

abcweizhuo, 2020-01-10:
## Can't build open62541 with UA_ENABLE_ENCRYPTION on win7
I try to build with CMake on Windows, with no UA_ENABLE_ENCRYPTION  CMake work well,
but with UA_ENABLE_ENCRYPTION   CMake  report error
## Background Information 
open62541 v1.0 
win7 64
CMake 3.16.2
VS2015

Used CMake options:
Commandline options:
-DBUILD_SHARED_LIBS:BOOL="1" -DUA_ENABLE_ENCRYPTION:BOOL="1" 

Cache file:
BUILD_SHARED_LIBS:BOOL=1
UA_ENABLE_ENCRYPTION:BOOL=1

error message: "Error in Configuration process, project files may be invalid."

ccvca, 2020-01-15:
You need to build mbedtls (Version 2.2.1, not the newest Version!) first and give the paths to open62541
https://github.com/ARMmbed/mbedtls/tree/mbedtls-2.2.1 (checkout tag "mbedtls-2.2.1").

Then you should be able to build it. (I've build it on Win 10 with the latest master-Versions)

abcweizhuo, 2020-01-19:
> You need to build mbedtls (Version 2.2.1, not the newest Version!) first and give the paths to open62541
> https://github.com/ARMmbed/mbedtls/tree/mbedtls-2.2.1 (checkout tag "mbedtls-2.2.1").
> 
> Then you should be able to build it. (I've build it on Win 10 with the latest master-Versions)

Thank you for the answer

closed 2020-01-19 by abcweizhuo

sirop, 2020-05-07:
@ccva @abcweizhuo 

If I have done everything correctly, mbedtls-2.2.1 produces only mbedTLS.lib, but no mbedX509.lib and no mbedcrypto.lib.
However, open62541 still awaits all three libs:  mbedTLS.lib, mbedX509.lib and mbedcrypto.lib.

What have I done wrong? Winsows 7, MSVC 2015...

sirop, 2020-05-07:
@ccva @abcweizhuo

False alarm, sorry. Everything was in the install directory.