open62541/open62541 #3353: Cannot connect to Softing dataFEED OPC UA Demo Server

jkahlert, 2020-01-10:
## Description

I tried to connect to the Softing dataFEED OPC UA Demo Server using the simple client from the client tutorial but the connection is rejected because of 'No suitable UserTokenPolicy' (see screenshot below). I think I have to modify the client configuration - but how do I do that?

## Background Information / Reproduction Steps
Source code (Microsoft Visual Studio 2017):

// ConsoleApplication9.cpp : Definiert den Einstiegspunkt für die Konsolenanwendung.
//

#include "stdafx.h"
#include <signal.h>
#include <stdio.h>

#include "open62541.h""

int main(void) {
	UA_Client *client = UA_Client_new();
	UA_ClientConfig_setDefault(UA_Client_getConfig(client));
//	UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:48010");
	UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://joergwin10:51510/UA/DemoServer");
	if (retval != UA_STATUSCODE_GOOD) {
		UA_Client_delete(client);
		getchar();
		return (int)retval;
	}
	/* Read the value attribute of the node. UA_Client_readValueAttribute is a
	* wrapper for the raw read service available as UA_Client_Service_read. */
	UA_Variant value; /* Variants can hold scalar values and arrays of any type */
	UA_Variant_init(&value);
	/* NodeId of the variable holding the current time */
	const UA_NodeId nodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_CURRENTTIME);
	retval = UA_Client_readValueAttribute(client, nodeId, &value);
	if (retval == UA_STATUSCODE_GOOD &&
		UA_Variant_hasScalarType(&value, &UA_TYPES[UA_TYPES_DATETIME])) {
		UA_DateTime raw_date = *(UA_DateTime *)value.data;
		UA_DateTimeStruct dts = UA_DateTime_toStruct(raw_date);
		UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "date is: %u-%u-%u %u:%u:%u.%03u\n",
			dts.day, dts.month, dts.year, dts.hour, dts.min, dts.sec, dts.milliSec);
	}
	/* Clean up */
	UA_Variant_clear(&value);
	UA_Client_delete(client); /* Disconnects the client internally */
	getchar();
	return EXIT_SUCCESS;
}

Console output:

[image]

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): Zip-file from Jan 06, 2020
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Win 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-01-14:
Did you enable encryption for the build?

jkahlert, 2020-01-15:
Yes, I did. I added 

#define UA_ENABLE_ENCRYPTION

to the open62541.h file and re-built the project (I'm not an experienced C programmer, normally I'm programming in DELPHI...). I have installed the MBEDTLS library from https://tls.mbed.org/download for encryption. It's no problem to connect to the server with the Demo UA Client from Softing as Anonymous.

Thanks in advance

jpfr, 2020-01-15:
You also have to configure encryption.
For example, you have to supply a client certificate and matching private key.

A certifiacte can be generated with the tools in /tools/cert.
See here for a code example: https://github.com/open62541/open62541/blob/master/examples/encryption/client_encryption.c

jkahlert, 2020-01-16:
I had seen this example before but could not get it running even after hours of work. I do not understand why I have to setup encryption if the server obviously supports a non-encrypted endpoint? With the Softing demo client I can connect as Anonymous without activating any encryption mechanism (see screenshot below). 
[image]
I have also tested the components from OPCLabs (they allow DELPHI programming) and had no problem to connect to the Softing demo server without any certificate or encryption, but I would prefer to use Open62541 if possible.

jkahlert, 2020-01-20:
In the meantime I could get it running (with all securityMode settings) - thanks!

closed 2020-02-13 by jpfr