open62541/open62541 #3359: Problems with prepending incomplete chunks for "lazy" network layers

andranik-s, 2020-01-13:
## Description

The code that prepends incomplete chunks for "lazy" network layers has issues.

The reason why the following code does not work
https://github.com/open62541/open62541/blob/d4285862e11db10e018c523ba60d79e97e8cd09d/src/ua_connection.c#L177-L182
are these lines
https://github.com/open62541/open62541/blob/d4285862e11db10e018c523ba60d79e97e8cd09d/src/ua_connection.c#L210
https://github.com/open62541/open62541/blob/d4285862e11db10e018c523ba60d79e97e8cd09d/src/ua_securechannel.c#L542-L543
because an incomplete message is freed before ```UA_SecureChannel_persistIncompleteMessages``` tries to copy it.

The issue was discovered while testing the WebSocket network layer. Before https://github.com/open62541/open62541/pull/3355 messages were not concatenated in the network layer which was resulting in errors of this type:
```
Invalid read of size 1
  at 0x5C1239E: Byte_decodeBinary.lto_priv.60 (ua_types_encoding_binary.c:204)
  by 0x5C1302B: NodeId_decodeBinary.lto_priv.66 (ua_types_encoding_binary.c:634)
  by 0x5C151B8: UA_decodeBinary (ua_types_encoding_binary.c:1505)
  by 0x5C01CC7: UA_NodeId_decodeBinary (types_generated_encoding_binary.h:248)
  by 0x5C03EE5: processMSG (ua_server_binary.c:638)
  by 0x5C04587: processSecureChannelMessage (ua_server_binary.c:733)
  by 0x5C19BFB: processMessage (ua_securechannel.c:560)
  by 0x5C19D98: UA_SecureChannel_processCompleteMessages (ua_securechannel.c:605)
  by 0x5C04B11: UA_Server_processBinaryMessage (ua_server_binary.c:889)
  by 0x5BA023D: callback_opcua (network_ws.c:201)
Address 0x96ae438 is 24 bytes inside a block of size 29,476 free'd
  at 0x4C30D3B: free (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
  by 0x5C11DD2: UA_Array_delete (ua_types.c:1206)
  by 0x5C0F16E: String_clear.lto_priv.94 (ua_types.c:142)
  by 0x5C11B32: UA_clear (ua_types.c:1135)
  by 0x5C17A96: UA_ByteString_deleteMembers.lto_priv.929 (types_generated_handling.h:470)
  by 0x5C1810C: UA_Connection_processChunks (ua_connection.c:210)
  by 0x5C04A62: UA_Server_processBinaryMessage (ua_server_binary.c:869)
  by 0x5BA023D: callback_opcua (network_ws.c:201)
Block was alloc'd at
  at 0x4C31D2F: realloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
  by 0x5C17FF6: UA_Connection_processChunks (ua_connection.c:182)
  by 0x5C04A62: UA_Server_processBinaryMessage (ua_server_binary.c:869)
  by 0x5BA023D: callback_opcua (network_ws.c:201)
```

## Background Information / Reproduction Steps

Comment out message concatenation from a network layer and send a reasonably big request as a client.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-01-14:
Thanks for the thorough background research.
This fix comes in #3358.

andranik-s, 2020-01-22:
@ccvca It's hard to tell.

The problem is that after these changes I'm experiencing issues with websockets: clients fail to decode some messages sent from the server. So I can't tell if this particular issue has been solved since I don't have working websockets.

I'll have to figure out what exactly is causing the issue before I can give an answer.

It might be the client code (third party), although it seemed to work ok before the PR.
It might be `network_ws.c`, which would not be surprising because it has plenty of other issues.

In short, I don't know. It might have solved the issue since the tcp layer seems to work fine with its chunk assembly code deleted.

andranik-s, 2020-01-22:
After some more fiddling I am pretty sure this issue has been fixed, but probably there are some new ones now with the websocket layer itself.

I will close this issue and open a new one if I am sure that it is a server side problem.

closed 2020-01-22 by andranik-s