open62541/open62541 #3360: can't make project

mojaxx, 2020-01-13:
hi
i am starting to use open62541 
i make the steps in issue #2567 but when i use make command (step 7 on issue) the progress stick at 88%
i try it several times 
please help me 

Sorry for my english.

thanks

Pro, 2020-01-13:
Try to use `-DCMAKE_BUILD_TYPE=Debug` which is building faster.

Release builds can take a few minutes with full namespace zero.

Please also read this:
https://open62541.org/doc/current/building.html

closed 2020-01-13 by Pro

mojaxx, 2020-01-14:
hi again 

thanks for that but i try and it did not progress more than 88% and i wait one hour 

i don't know what to do 

thanks
[image]

mojaxx, 2020-01-14:
hi  again
thanks very much for your help
sorry but i have one more problem
at the end when i want to start the server it gives an error as shown in blew figure
i appreciate if you help me 
thanks a lot
[image]

mojaxx, 2020-01-14:
Hi
Thanks very much