open62541/open62541 #3361: ziptree.h is not included in building RT publisher

ZekeLandon, 2020-01-14:
Hi,

I am trying to build RT publisher as mentioned in this [link](https://github.com/open62541/open62541/tree/master/examples/pubsub_realtime#building-the-rt-publisher). Build fails as ziptree.h location is unknown to the compiler.

<pre><font color="#8AE234"><b>admin@UbuntuPC</b></font>:<font color="#729FCF"><b>~open62541/build_rt</b></font>$ gcc -O2 ../examples/pubsub_realtime/pubsub_interrupt_publish.c ../examples/pubsub_realtime/bufmalloc.c -I../include -I../plugins/include -Isrc_generated -I../arch/posix -I../arch -I../plugins/networking bin/libopen62541.a -lrt -o rt_publisher
In file included from <b>../include/open62541/server_config.h:15:0</b>,
                 from <b>../plugins/include/open62541/server_config_default.h:13</b>,
                 from <b>../examples/pubsub_realtime/pubsub_interrupt_publish.c:13</b>:
<b>../include/open62541/plugin/nodestore.h:21:10:</b> <font color="#EF2929"><b>fatal error: </b></font>ziptree.h: No such file or directory
 #include <font color="#EF2929"><b>&quot;ziptree.h&quot;</b></font>
          <font color="#EF2929"><b>^~~~~~~~~~~</b></font>
compilation terminated.</pre>

ziptree.h header file is present in deps folder which appears to be home directory of stack. Hence, the build step should be modified as follows.

gcc -O2 ../examples/pubsub_realtime/pubsub_interrupt_publish.c ../examples/pubsub_realtime/bufmalloc.c **-I../deps** -I../include -I../plugins/include -Isrc_generated -I../arch/posix -I../arch -I../plugins/networking bin/libopen62541.a -lrt -o rt_publisher

Is that right?

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.1-dev-289-gd02fc4f5
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 18.04 LTS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-01-14:
That's right.

ZekeLandon, 2020-01-14:
@jpfr Then I will close the issue for now. Please update the documentation whenever possible.

Regards,
Zeke

closed 2020-01-14 by ZekeLandon