open62541/open62541 #3362: Alarms & Conditions in open62541

OPCUAUser, 2020-01-14:
Hello Community,

I  need to implement Alarms & Conditions in my system and I believe open62541 stack has that source code of the feature. I wish to know that the latest stack is a complete source or is it still a work-in-progress item?

Thanks in advance!

ccvca, 2020-01-14:
An experimental option is available (#2083), which might be enought for your implementation.

OPCUAUser, 2020-01-20:
@ccvca Thanks for the information. I hope this helps for my implementation.

jpfr, 2020-01-20:
No, we are not feature complete for Events.
But, afaik, no SDK vendor is complete. The filter mechanism can get quite complicated.

Counter-question. Which features do you need?

mtbeck, 2020-01-30:
I'm in the same situation as @OPCUAUser. Does this stack support Alarms & Condition, conditiontype "OffNormal"?

sqadasi, 2020-01-30:
> 
> 
> I'm in the same situation as @OPCUAUser. Does this stack support Alarms & Condition, conditiontype "OffNormal"?

The supported Parts of A&C are listed in this comment:)
https://github.com/open62541/open62541/pull/2083#issue-217690330

OPCUAUser, 2020-01-31:
@jpfr : Thanks for your response. To be very sure, I see from the latest open62541 SDK that the A&C base condition server, Enable, Address space instance, Alarm server and Acknowledgeable Alarm server Facets are initially implemented. For now, it is enough for my basic tests. Is there a roadmap to implement other Facets (Dialog, Refresh2, etc) as well?

Thanks

jpfr, 2020-02-14:
Additional A&C features depend on developers to step up and work on the topic.

Please reopen if required.

closed 2020-02-14 by jpfr