open62541/open62541 #3364: Use open62541 with Encryption on Windows without MbedTLS on system path

ccvca, 2020-01-14:
## Description
After 441ac87bfac31c8d10e126d1e373e92b622dbde3 it's quite hard to build with encryption, when mbedtls is not installed on a system path (e.g. on windows), as the targets that are defined by FindMbedTLS are not exported. (mbedtls, mbedx509, mbedcrypto)

https://github.com/open62541/open62541/blob/d4285862e11db10e018c523ba60d79e97e8cd09d/tools/cmake/FindMbedTLS.cmake#L22

Using the open62541 library in an own project causes that cmake tries to link to the libraries by name, as the target is no longer known. This only works, if the location of libmbedtls.a (...) is on a well known library path.

## Background Information / Reproduction Steps
Build + Install open62541 with Encryption by cmake

Use find_package(open62541 REQUIRED) in an own project

## Possible Solutions
1. Export the targets from FindMbedTLS/ revert 441ac87bfac31c8d10e126d1e373e92b622dbde3 
    - Disadvantage: Absolute paths are bad e.g. when building the deb-Package
2. Update documentation that FindMbedTLS must be used additional to find_package(open62541 REQUIRED) when using encryption
    - Disadvantage: Hard to do, as FindMbedTLS  not on the default seach path
3. include FindMbedTLS in [open62541Config.cmake.in](https://github.com/open62541/open62541/blob/d4285862e11db10e018c523ba60d79e97e8cd09d/tools/cmake/open62541Config.cmake.in)
4. Link static and include it into the open62541 library
    - Disadvantage: I don't know it this is expected by mbedTLS, might require building mbedTLS by open62541 from source

My favorite is the 3rd one, but I didn't find any cmake-examples that have done this in the past...

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): d4285862e11db10e018c523ba60d79e97e8cd09d
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jadhavshital, 2020-02-18:
You can use this link to build mbedtls.lib (https://tls.mbed.org/kb/compiling-and-building/using-mbedtls-in-microsoft-visual-studio-2015)
And after that set encryption flag on.
Also set following flags -
1. MBEDCRYPTO_LIBRARY - <your_path>\mbedtls-2.16.4-apache\mbedtls-2.16.4-apache\mbedtls-2.16.4\visualc\VS2010\Release
2. MBEDTLS_INCLUDE_DIRS - <your_path>\mbedtls-2.16.4-apache\mbedtls-2.16.4-apache\mbedtls-2.16.4\include
3. MBEDTLS_LIBRARY - <your_path>\mbedtls-2.16.4-apache\mbedtls-2.16.4-apache\mbedtls-2.16.4\visualc\VS2010\Release
4. MBEDX509_LIBRARY - <your_path>\mbedtls-2.16.4-apache\mbedtls-2.16.4-apache\mbedtls-2.16.4\visualc\VS2010\Release

I have built open62541 lib successfully on windows 10.

ccvca, 2020-02-18:
@jadhavshital Thanks for oyur anwer. I was also able to build the library with encryption support.

The problem occures when open62541 is integrated in a project. As the open62541-library links to the cmake-target mbedtls and this target is not exported, linking open62541 by find_package() causes cmake to link to a library mbedtls on the library path (does not use the original mbedtls-Target anymore).

ccvca, 2020-06-17:
Closed by #3714

closed 2020-06-17 by ccvca