open62541/open62541 #3365: Cannot compile the example tutorial_pubsub_publish and tutorial_pubsub_subscribe

BreemTUKL, 2020-01-14:
## Description

Hello everybody,

I am currently trying to run the examples tutorial_pubsub_publish.c and  tutorial_pubsub_subscribe.c
At tutorial_pubsub_publish.c I wrote this:
gcc -std = c99 -o PubSub_Publish_Example open62541.c tutorial_pubsub_publish.c (how can I know if everything is compiled correctly here? nothing happens)

[image]

And then when I run tutorial_pubsub_subscribe.c I am also  take:
gcc -std = c99 -o PubSub_Publish_Example open62541.c tutorial_pubsub_subscribe.c
but I get the error message here:
tutorial_pubsub_subscribe.c: 19: 10: fatal error: ua_pubsub.h: No such file or directory. 19 | #include "ua_pubsub.h" compilation terminated.

[image]

I have activated the option UA_ENABLE_AMALGAMATION = ON and
I added the files open62541.c and open62541.h in the /open62541/example/pubsub folder. Then I deactivated the option UA_ENABLE_AMALGAMATION = OFF.
Is that right what I did?

Can somebody help me please?
Thanks in advance
Breem

## Background Information / Reproduction Steps
cmake -DBUILD_SHARED_LIBS=OFF -DCMAKE_BUILD_TYPE=Debug -DUA_ENABLE_AMALGAMATION=OFF -DUA_BUILD_EXAMPLES= ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_ETH_UADP=ON -DUA_NAMESPACE_ZERO=FULL  ..

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 19.10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

maleselo, 2020-01-14:
You should follow these instructions here: https://open62541.org/doc/1.0/building.html#building-the-library

```
cd open62541
mkdir build
cd build
cmake .. -DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_PUBSUB=ON
make
```

Then on ./build/bin/examples you should retrieve pubsub tutorial executables.

BreemTUKL, 2020-01-14:
Hello,
@maleselo Thank you for your response.

I already took the steps right at the beginning.
my problem was the executable tutorials.
I thought I would run the code in the sample folder:
gcc -std = c99 -o PubSub_Publish_Example open62541.c tutorial_pubsub_publish.c
and
gcc -std = c99 -o PubSub_Publish_Example open62541.c tutorial_pubsub_subscribe.c

and then run with ./tutorial_pubsub_publish.c and ./tutorial_pubsub_subscribe.c in the same example folder.

but now i see that in open62541/build/bin/examples and the files should be executed with "./" And that was my problem.

That means as soon as you activate the option UA_BUILD_EXAMPLES = ON, the examples are compiled in the folder open62541/build/bin/examples and I can run directly from there with "./" without executing gcc -std = c99 -o beforehand, is that's the right?
Did I need the open62541.c and open62541.h ? when i want to run examples?
best regards
Breem

maleselo, 2020-01-14:
@BreemTUKL 

tutorial_pubsub_publish.c and tutorial_pubsub_subscribe.c are source files, there are not executable.

[gcc](https://gcc.gnu.org/) is a compiler, a program which basically converts source code into executable.

Maybe you should make yourself a little bit more comfortable with C compilation process and build tools like Makefile and CMake before going futher ;)

Here some ressources you can start with:
https://www.calleerlandsson.com/the-four-stages-of-compiling-a-c-program/
https://www3.ntu.edu.sg/home/ehchua/programming/cpp/gcc_make.html

Regards,
Maleselo

BreemTUKL, 2020-01-14:
Hello @maleselo,
ok thank you for your help and tips :-)
best regards.

closed 2020-01-14 by BreemTUKL

maleselo, 2020-01-15:
@BreemTUKL 

No worries, I thought you had no clue what is gcc and compilation at first but then I realized that was just typo mistakes ;) 

Regards,
Maleselo