open62541/open62541 #3366: When to support UA client multithreading

fengshaoxie, 2020-01-14:
## Description
When to support UA client multithreading
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

fengshaoxie, 2020-01-15:
[image]

teccheck, 2025-04-28:
I'm not quite sure what you mean. Multithreading ist supported in recent versions via the `UA_MULTITHREADING` Cmake flag. See https://www.open62541.org/doc/master/building.html for more information. Feel free to reopen this issue and clarify if you want

closed 2025-04-28 by teccheck