open62541/open62541 #3368: Cannot Compile tutorial_server_historicaldata.c in examples

mojaxx, 2020-01-14:
HI 
I am trying to compile this example but it gives me this error :server.c:9:70: fatal error: open62541/plugin/historydata/history_data_backend_memory.h: No such file or directory
 #include <open62541/plugin/historydata/history_data_backend_memory.h>
                                                                      ^
compilation terminated.

Do i need to install or do anything before compiling this example ??
thanks

maleselo, 2020-01-14:
@mojaxx You may need to enable UA_ENABLE_EXPERIMENTAL_HISTORIZING option

mojaxx, 2020-01-15:
HI 
Thanks for that 
Where should i enable this??
Thanks

mojaxx, 2020-01-15:
Thank you a lot
In general do i need to install data base on beaglebone black or what?

mojaxx, 2020-01-18:
Hi
There is no opinion??

closed 2020-01-18 by mojaxx