open62541/open62541 #3377: Decoding of UA_Variant / ExtensionObject containing custom DataType

KuabeM, 2020-01-16:
## Description

I try to decode a custom data type `KeyIntPair` which I'm receiving in a `UA_Variant`. As described in the [docs][variant-doc], the type is wrapped in an `ExtensionObject` and I'm able to extract that via an `reinterprete_cast<ExtensionObject*>(input->data)`. I also know that it's encoded as `UA_EXTENSIONOBJECT_ENCODED_BYTESTRING` but I'm not able to extract the actual `KeyIntPair` from the `content.encoded.body`. Here's what I tried:

```
UA_Variant* input;
...
UA_ExtensionObject* eo = reinterpret_cast<UA_ExtensionObject*>(input->data);
UA_KeyIntPair* kp = UA_KeyIntPair_new();
kp = reinterpret_cast<UA_KeyIntPair*>(eo->content.encoded.body.data); \\ data of kp is not accessible
UA_KeyIntPair_decodeBinary(&eo->content.encoded.body, 0, kp); \\ can't import header 
```

I'm using the nodeset compiler to generate the code. In the `types_{namespace}_generated_encoding_binary.h` header are corresponding `UA_KeyIntPair_encodeBinary` and `decodeBinary` functions, but when I try to import that header, the compiler can't find the `ua_types_encoding_binary.h`. In #2919, you also say that one shouldn't use the encode and decode methods but 'let the stack handle it'. Unfortunately, you only give an example for encoding, but not the decoding. I've also seen #1420, but I think that's also the other way round.

So can you please give me a hint how to do the decoding or as an alternative, how I can include the `encoding_binary.h` header? Thanks.

## Background Information / Reproduction Steps

I set-up the nodeset and model.xml with the UaModeler and connect to the server with the UaExpert. The resulting struct `KeyIntPair` is generated as
```
typedef struct {
    UA_String key;
    UA_Int32 value;
} UA_KeyIntPair;
```

Used CMake options:

Compiled open62541 with
```
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/home/XX/.local -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_ENCRYPTION=ON UA_NAMESPACE_ZERO=FULL
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

[variant-doc]: https://open62541.org/doc/1.0/types.html#variant

andranik-s, 2020-01-29:
If I understand correctly what you are trying to do, your ExtensionObject should already be decoded.
So I think you should do something like this:
```
UA_ExtensionObject* eo = reinterpret_cast<UA_ExtensionObject*>(input->data);
if(eo->encoding == UA_EXTENSIONOBJECT_DECODED) {
  kp = reinterpret_cast<UA_KeyIntPair*>(eo->content.decoded.data);
  ...
}
```
Take a look at how `UA_HistoryReadRequest::historyReadDetails` extension object is handled here:
https://github.com/open62541/open62541/blob/359f56e40dea85d69b1008184c1fd309779e1ac7/src/server/ua_services_attribute.c#L1561-L1576

<sub>Side note: It's ok (preferred actually) to use `static_cast` for `void*`</sub>

KuabeM, 2020-01-29:
Thanks for your answer. So far, your understanding is correct. However, the encoding of the Extensionobject is `UA_EXTENSIONOBJECT_ENCODED_BYTESTRING` and not `.._DECODED`. The suggested example just fails `UA_STATUSCODE_BADNOTSUPPORTED`. How is the encoding set/influenced here? As an alternative, how can I decode the bytestring?

Some more Background: we're upgrading the library from release 0.3 to 1.0, with the old release your way is working and the `eo` is already decoded. Additionally, In the code which is calling the method I'm explicitly setting the encoding to `decoded`, so that should be fine right?
<details>
  <summary> It looks like that</summary>

```c
    UA_String argString = UA_String_fromChars(const_cast<char*>(key.c_str()));
    UA_KeyIntPair my_value;
    my_value.key = argString;
    my_value.value = value;

    UA_ExtensionObject eo;
    UA_ExtensionObject_init(&eo);
    eo.encoding = UA_EXTENSIONOBJECT_DECODED;
    eo.content.decoded.data = &my_value;
    eo.content.decoded.type = &OPC_UA_SERVICE_TYPES[OPC_UA_SERVICE_TYPES_KEYINTPAIR];

    UA_Variant input;
    UA_Variant_init(&input);
    UA_Variant_setScalarCopy(&input, &eo, &UA_TYPES[UA_TYPES_EXTENSIONOBJECT]);

    UA_Variant *output;
    size_t outputSize;
    UA_StatusCode retval = UA_Client_call(client, object_id, replace_id, 1, &input, &outputSize, &output);
```
And I also tried it with UaExpert, but it gives exactly the same result...
 </details>

andranik-s, 2020-01-30:
I've never used custom data types, but looking at the decoder's code I can tell that extension objects won't be decoded if the wrapped object's type is not recognised: (line 885)
https://github.com/open62541/open62541/blob/8cecdb5cffb3ff15737685f679d03c019c67d9e9/src/ua_types_encoding_binary.c#L878-L885
`UA_findDataTypeByBinaryInternal` searches for the type in `UA_TYPES` and in custom data types.

Again I've never used any custom types but I guess they should be set in server's and client's config:
https://github.com/open62541/open62541/blob/8cecdb5cffb3ff15737685f679d03c019c67d9e9/include/open62541/client_config.h#L98
https://github.com/open62541/open62541/blob/8cecdb5cffb3ff15737685f679d03c019c67d9e9/include/open62541/server_config.h#L116

andranik-s, 2020-01-30:
Oh! There are examples actually
Take a look at open62541/examples/custom_datatype

andranik-s, 2020-01-30:
In short, make shure you've set custom types in both client's and server's config, and use those types directly without manully using extension objects:
```
UA_Variant_setScalarCopy(&input, &my_value,
                         &OPC_UA_SERVICE_TYPES[OPC_UA_SERVICE_TYPES_KEYINTPAIR]);
```

KuabeM, 2020-02-03:
I've seen the example already and I do add the custom data types to the server. I tried all the suggested things at some point, but still no luck.. I hoped that I was just missing a simple step.

Referring to my original question, what is the purpose of the `types_{namespace}_generated_encoding_binary.h` file and its `UA_{type}_decodeBinary()` methods? When I include the `ua_types_encoding_binary.c` header in order to use these methods, I can successfully decode the KeyIntPair values.

tuxroger, 2020-02-11:
I am having the same problem.
I have a custom data type which is a struct:
typedef struct {
	UA_String uuid_;
	UA_String customerCode_;
	UA_String internalErrorCode_;
	UA_String file_;
	UA_UInt32 line_;
} SystemError;

When I retrieve the node from the server, I get it a binarystring, which I captured in wireshark:
string1
05 00 00 00      // len
75 75 69 64 31 // data
string2
09 00 00 00      // len
63 75 73 74 6f 6d 65 72 31 // data
string3
09 00 00 00     // len
69 6e 74 65 72 6e 61 6c 31 // data
string4 
05 00 00 00       // len
66 69 6c 65 31 // data
uint32
45 00 00 00   // data

I have been trying to search a function in the sdk that helps me decode basic data types and all the functions I have found are internal and cannot be used from the library.
Thanks for your time.

jpfr, 2020-02-12:
As the remote side also open62541 or another SDK?
In open62541 you can set custom data types in the client/server config.
There are also examples for that.

tuxroger, 2020-02-12:
Hi,
the remote side is a Siemens PLC. I have created custom data types (named UDTs in Siemens environment), so I have almost no control in the way the data is published.
I attach a picture of the wireshark capture so you can see what I get on the wire.

[image]

At least, using the open62541 stack I get the stream of bytes containing the data of the struct mentioned before, but now I would like to know if there are already builtin decoding functions for all the basic UA types such string, int, etc so that I can easily retrieve the data one by one from the stream.
If that is not the case, I'll write my own decode functions for every UA item I need to decode.
Thanks for your time and for your quick response!

tuxroger, 2020-02-12:
Hi,
Actually, I found the funcions for encoding/deconding binary streams but they are not accessible using either the amalgamation header or the header files distributed regularly.
Do you plan to expose that funcions at some point?
Thanks!

KuabeM, 2020-02-13:
In my case, server is open62541 and on the client side i tried both open62541 and UaExpert.

tuxroger, 2020-02-14:
Hi,
In my case, I have a Siemens PLC with a built-in OPCUA server and on the client side, I am writing a data logging application using the open62541 stack. What I am missing is the helping funcions for encoding / decoding the binary data I get from the PLC.
I have seen them in the ua_encoding_binary.h header file, but in the amalgamation they are not exposed in the header file.
I was asking the dev team if they were thinking to expose that functions in another module like tools.h or something similar so that people can use them.
Thanks!
Roger.

jpfr, 2020-02-14:
@tuxroger , you can use the "normal" source tree instead of the single-file release.
Then you can use the "private" internal API.
That should work quite well. But we make no guarantee for private API to be stable over time.

tuxroger, 2020-06-18:
Thanks!
Sorry for the late reply :)
Roger

jpfr, 2021-09-23:
The binary encoding api is now public.

closed 2021-09-23 by jpfr