open62541/open62541 #3381: Working with objects

preone, 2020-01-17:
## Description
I'm trying to manually update an attribute value inside an object. For example, in the tutorial_server_object.c, how can i change the value of motor rpm from 50 to 60? I'm trying to use findSinglechildnode with no luck.
In my project I'm using callbacks. 

Thank you!

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: openwrt
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

juangburgos, 2020-01-21:
Use the `UA_Server_writeValue` function.

preone, 2020-01-22:
To use that function, you should point the right nodeId. In the case of the example, the nodeId is numeric and generated randomly. My problem is how to get the numeric nodeid of the attribute inside an object after the creation.
I solved using string reference instead of numeric reference, but I would be grateful if someone can help me to understand how to get the numeric nodeId of an attribute/object.
Thank you for reply

juangburgos, 2020-01-22:
When you have a custom type and create an insteance of it, indeed the children variables and objects get random nodeIds. The only nodeId that you know for sure is the parent nodeId, so what you have to do is **browse** from the parent to the child using the library's browse API (`UA_Server_browse`). 

For example you can browse forward for hierachical references, and then loop the result list until a child node matches the browse name of the child you are looking for.

preone, 2020-01-22:
Thank you so much!

closed 2020-01-22 by preone