open62541/open62541 #3385: Undisplay Info/Session

lenfep, 2020-01-20:
## Description

How to undisplay the info/session in terminal? Is it configured when build and install the library?

jpfr, 2020-01-20:
It is configured during build.
And you can define a custom logger plugin that selects which logs to actually print.

lenfep, 2020-01-21:
how you configure custom logger when build ? i use cmake  -DCMAKE_BUILD_TYPE=MinSizeRel -DUA_NAMESPACE_ZERO=MINIMAL  ..

juangburgos, 2020-01-21:
If for the server, take a look at the `UA_ServerConfig` definition:

```c
struct UA_ServerConfig {
    UA_UInt16 nThreads; /* only if multithreading is enabled */
    UA_Logger logger;

// ...

}
```

You need to overwrite the `UA_Logger logger` which consist of setting the function pointers to your own callbacks:

```c
typedef struct {
    /* Log a message. The message string and following varargs are formatted
     * according to the rules of the printf command. Use the convenience macros
     * below that take the minimum log-level defined in ua_config.h into
     * account. */
    void (*log)(void *logContext, UA_LogLevel level, UA_LogCategory category,
                const char *msg, va_list args);

    void *context; /* Logger state */

    void (*clear)(void *context); /* Clean up the logger plugin */
} UA_Logger;
```

Basically, just browse the source code, man...

closed 2020-01-25 by jpfr