open62541/open62541 #3387: Server, memory leak after deleting parent with lots of children

juangburgos, 2020-01-20:
## Description & Reproduction Steps

I don't know if I am missing something here, maybe I am using the API wrong. Or maybe the behaviour is as intended.

I have simplified the issue to a small example where I create a parent node and then add a lot of children to it. Then delete the parent node.

```c
#include <open62541.h>
#include <signal.h>

static UA_NodeId createParent(UA_Server *server) 
{
    UA_NodeId parentId; 
    UA_ObjectAttributes oAttr = UA_ObjectAttributes_default;
    oAttr.displayName = UA_LOCALIZEDTEXT("en-US", "Parent");
    UA_Server_addObjectNode(server, UA_NODEID_NULL,
                            UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                            UA_QUALIFIEDNAME(1, "Parent"), UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE),
                            oAttr, NULL, &parentId);
    return parentId;
}

static void addManyChildrenToParent(UA_Server* server, const UA_NodeId parentId, const int numChildren)
{
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.displayName = UA_LOCALIZEDTEXT("en-US", "Child");
    UA_Int32 myInteger = 1;
    UA_Variant_setScalar(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
    for (int i = 0; i < numChildren; i++)
	{
		UA_Server_addVariableNode(server, UA_NODEID_NULL, parentId,
			UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
			UA_QUALIFIEDNAME(1, "Child"),
			UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), attr, NULL, NULL);

    }
}

static void deleteParent(UA_Server* server, const UA_NodeId parentId)
{
    UA_Server_deleteNode(server, parentId, true);
}

UA_Boolean running = true;
static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    UA_NodeId parentId = createParent(server);
    addManyChildrenToParent(server, parentId, 10000);
    deleteParent(server, parentId);

    UA_StatusCode retval = UA_Server_run(server, &running);

    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

I first benchmark the example by commenting the `addManyChildrenToParent` line and see how much memory the server is using after running for some seconds. Looking at the Visual Studio *Diagnostic Tools*, it shows `1.3MB`:

[image]

Then I uncomment the line and get the the server is using now `2.5MB`.

[image]

Creating a couple of memory snapshots before and after creating and deleting the parent, I can see the extra memory is due to `UA_ReferenceTarget` allocations:

[image]

In the `addReferenceTarget` in the following line:

[image]

I don't know if this is on purpose, maybe is some mechanism to reuse allocated memory or something. I am just wondering because I have an application where many nodes are created and deleted dynamically and see the memory growing and growing as the user creates and removes lots of nodes with lots of children.

Thank you for your help.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): tag 1.0
 - [x] Other OPC UA SDKs used (client or server): none
 - [x] Operating system: Windows
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached : N/A
 - [x] Wireshark network dump attached: N/A
 - [x] Self-contained code example attached: yes
 - [x] Critical issue : maybe?

damijans, 2020-01-22:
Hi,
I have checked your example on Linux, however I was not able to reproduce reported issue. i.e. after deleting parent node in deleteParent() function memory was correctly freed. Would you mind to specify what kind of options (UA_xxx) did you use for compiling open62451.

pls, don't expect solutions from me, I don't work professionally on this project, it's just my curiosity.

juangburgos, 2020-01-22:
No prob, I copiled with;

```
-DUA_ENABLE_AMALGAMATION=ON -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON
```

juangburgos, 2020-01-22:
Please note that all memory is correctly freed after calling `UA_Server_delete(server);`. But I would exect that the memory is already cleaned after calling `deleteParent(server, parentId)`.

jpfr, 2020-01-30:
Fixed via #3407.

closed 2020-01-30 by jpfr