open62541/open62541 #3391: server_ctt usage with certificates

kjeldflarup, 2020-01-21:
## Description server_ctt usage with certificates
server_ctt has three command line parameters (linux version)
```
--trustlistFolder <folder>
--issuerlistFolder <folder>
--revocationlistFolder <folder>
```
It is my understanding that this can be used to add trusted certificates to a server, but documentation on this is missing.

- What kind of files should go into each folder?
- Which format should the files have (standard)?
- How is the files generated?
- Which tools can be used to demo the functionality?

## Background Information / Reproduction Steps
I want to set up a framework, where the clients are authorized using their public key, like it is done with ex. SSH

 - [ ] open62541 Version (release number or git tag): 1.0
 - [ ] Other OPC UA SDKs used (client or server): UAExpert
 - [ ] Operating system: Linux

juangburgos, 2020-01-21:
Maybe [this link helps](https://github.com/juangburgos/QUaServer#create-certificates).

kjeldflarup, 2020-01-23:
Thanks Juan
That took me forward. I however still ends up in this error on the server
```
[2020-01-23 12:46:47.317 (UTC+0100)] info/network	Connection 5 | Processing the message failed with error BadCertificateUntrusted
```

Now my server is not QT but `server_ctt.c`, I can see that You set these for Your server
```
server.setApplicationName ("my_app");
server.setApplicationUri  ("urn:juangburgos:my_app");
server.setProductName     ("my_product");
server.setProductUri      ("juangburgos.com");
server.setManufacturerName("My Company Inc.");
server.setSoftwareVersion ("6.6.6-master");
server.setBuildNumber     ("gvfsed43fs");
```
server_ctt.c only sets this
```
    config.applicationDescription.applicationUri =
        UA_String_fromChars("urn:open62541.server.application");
```

A least this calls for removing `juangburgos` from generation of the certificates. 
```
openssl req -new -x509 -days 3600 -key ca/ca.key -subj "/CN=open62541.server.application CA/O=open62541.server.application" -out ca/ca.crt
```

But I'm also in doubt of this
```
openssl req -new -sha256 \
-key server/server.key \
-subj "/C=ES/ST=MAD/O=MyServer/CN=localhost" \
-out server/server.csr
```

Finally I do this to start server_ctt
```
$ cp -p ./server/server.crt.der ./trust/server.crt.der
$ server_ctt server/server.crt.der server/server.key.der --trustlistFolder trust
[2020-01-23 13:13:23.428 (UTC+0100)] warn/userland	No CA trust-list provided. Any remote certificate will be accepted.
[2020-01-23 13:13:23.434 (UTC+0100)] info/server	Reloading the trust-list
[2020-01-23 13:13:23.434 (UTC+0100)] info/server	Loaded certificate from trust
[2020-01-23 13:13:23.442 (UTC+0100)] warn/server	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2020-01-23 13:13:23.442 (UTC+0100)] warn/server	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2020-01-23 13:13:23.442 (UTC+0100)] warn/server	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2020-01-23 13:13:23.442 (UTC+0100)] info/network	TCP network layer listening on opc.tcp://kfa-Latitude-5580:4840/
```

I guess that this should be ignored `No CA trust-list provided. Any remote certificate will be accepted.`  
The next lines shows that server_ctt actually reads a trust list.

But what does this warning mean?
```
The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy 
```

When I try to connect from UAExpert the server writes this:
```
[2020-01-23 13:18:48.532 (UTC+0100)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[2020-01-23 13:18:48.533 (UTC+0100)] info/channel	Creating a new SecureChannel
[2020-01-23 13:18:48.533 (UTC+0100)] warn/securitypolicy	Security policy None is used to create SecureChannel. Accepting all certificates
[2020-01-23 13:18:48.533 (UTC+0100)] info/channel	Connection 5 | SecureChannel 5 | Opened SecureChannel
[2020-01-23 13:18:48.533 (UTC+0100)] info/channel	Connection 5 | SecureChannel 5 | CloseSecureChannel
[2020-01-23 13:18:48.534 (UTC+0100)] info/network	Connection 5 | Closed
[2020-01-23 13:18:48.552 (UTC+0100)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[2020-01-23 13:18:48.553 (UTC+0100)] info/channel	Creating a new SecureChannel
[2020-01-23 13:18:48.553 (UTC+0100)] info/server	Reloading the trust-list
[2020-01-23 13:18:48.553 (UTC+0100)] info/server	Loaded certificate from trust
[2020-01-23 13:18:48.553 (UTC+0100)] info/network	Connection 5 | Processing the message failed with error BadCertificateUntrusted
[2020-01-23 13:18:48.553 (UTC+0100)] info/network	Connection 5 | Closed
```

I don't know if this warning matters `warn/securitypolicy	Security policy None is used to create SecureChannel. Accepting all certificates`  
but it seems that it tries to connect twice.

juangburgos, 2020-01-23:
Did you include your `urn:open62541.server.application` in the `subjectAltName` part of your `exts.txt` file when generating the certificates?

Also, what error does UaExpert shows?

Checkout [this issue](https://github.com/juangburgos/QUaServer/issues/18). Also if you get stucked, just search for the `The configured ApplicationURI does not match the URI` text in the open62541 source code. The good thing about the C language is that it is very easily readible and maybe you cna figure out the problem.

kjeldflarup, 2020-01-23:
Adding `urn:open62541.server.application` to `exts.txt` removed this warning `The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256`

But still untrusted like above.

Here is the output of UAExpert
```
15:02:09.496 | Server Node        | localcert                      | Endpoint: 'opc.tcp://localhost:4840/
'
15:02:09.496 | Server Node        | localcert                      | Security policy: 'http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256'
15:02:09.496 | Server Node        | localcert                      | ApplicationUri: 'urn:open62541.server.application'
15:02:09.496 | Server Node        | localcert                      | Used UserTokenType: Anonymous
15:02:09.513 | General            |                                | [uastack] OpcUa_TcpConnection_ProcessResponse: Error Message!
15:02:09.514 | General            |                                | [uastack] OpcUa_TcpConnection_ProcessResponse: Status 0x80130000!
15:02:09.514 | Server Node        | localcert                      | Error 'BadSecurityChecksFailed' was returned during OpenSecureChannel
15:02:09.514 | Server Node        | localcert                      | Connection status of server 'localcert' changed to 'Disconnected'.
```

The rest of the application settings BTW follow source defaults
```
#define MANUFACTURER_NAME "open62541"
#define PRODUCT_NAME "open62541 OPC UA Server"
#define PRODUCT_URI "http://open62541.org"
#define APPLICATION_NAME "open62541-based OPC UA Application"
#define APPLICATION_URI "urn:unconfigured:application"
#define APPLICATION_URI_SERVER "urn:open62541.server.application"
```

juangburgos, 2020-01-23:
I have not yet figured out how to make the server support client login using certificates, haven't had the need so far. The only thing I support in my project `QUaServer` is for the client to trust the server, not the other way around. So this is as far as I can help you. Maybe try to suceed first on making the client trust the server using encryption and simple username-password login, then try to extrapolate that knowledge to make it work the other way around.

Good luck.

kjeldflarup, 2020-01-24:
OK I think that I got some progress now.  
Together with the code which generated this error `Processing the message failed with error BadCertificateUntrusted`, i found some out commented debug code to write out the mbedtls error
```
Verifying the certificate failed with error: The certificate is not correctly signed by the trusted CA
```
That made it obvious, that if the certificate used by UAExpert made made with the CA just generated, it would be accepted. That is still a todo.

Looking at the code in  https://github.com/open62541/open62541/blob/5dce2701a3156b1489fd6661deb150861fc082e1/plugins/ua_pki_default.c#L197 I found that another authentication is possible. 
```
cp $HOME/.config/unifiedautomation/uaexpert/PKI/own/certs/uaexpert.der trust/
```

Then the certificate is accepted (and I was rejected for not giving username and password.)  

But I will try with the CA option again next week. It makes much more sence just to install a single trusted CA on the target than to install a zillion client certificates.

andreasebner, 2025-02-04:
This issue seems to be outdated as the ctt example is not public for the latest releases. If there are known issues, feel free to open a new issue.

closed 2025-02-04 by andreasebner