open62541/open62541 #3394: server_ctt with Enable Anonymous error

kusumy, 2020-01-24:
## Description
I'm trying to execute server_ctt example (on Raspberry PI) with Anonymous option enabled, using following command:

```bash
./server_ctt --enableAnonymous
```

But there is error message said:

```bash
Unable to load file --enableAnonymous
```

Is my command to execute server_ctt with ananymous enabled is correct ?

## Background Information / Reproduction Steps

I use source file from master.zip 

https://github.com/open62541/open62541/archive/pack/master.zip

and build the source on Raspeberry PI with the following options:

```bash
 UA_ARCHITECTURE                  posix                                                                                                                                                                                                                                                                                                                  
 UA_BUILD_EXAMPLES                OFF                                                                                                                                                      
 UA_BUILD_FUZZING                 OFF                                                                                                                                                      
 UA_BUILD_FUZZING_CORPUS          OFF                                                                                                                                                      
 UA_BUILD_OSS_FUZZ                OFF                                                                                                                                                      
 UA_BUILD_TOOLS                   OFF                                                                                                                                                      
 UA_BUILD_UNIT_TESTS              OFF                                                                                                                                                      
 UA_COMPILE_AS_CXX                OFF                                                                                                                                                      
 UA_DEBUG                         OFF                                                                                                                                                      
 UA_DEBUG_DUMP_PKGS               OFF                                                                                                                                                      
 UA_ENABLE_AMALGAMATION           ON
 UA_ENABLE_COVERAGE               OFF                                                                                                                                                      
 UA_ENABLE_DA                     ON                                                                                                                                                       
 UA_ENABLE_DETERMINISTIC_RNG      OFF                                                                                                                                                      
 UA_ENABLE_DISCOVERY              ON                                                                                                                                                       
 UA_ENABLE_DISCOVERY_MULTICAST    OFF                                                                                                                                                      
 UA_ENABLE_DISCOVERY_SEMAPHORE    ON                                                                                                                                                       
 UA_ENABLE_ENCRYPTION             OFF                                                                                                                                                      
 UA_ENABLE_EXPERIMENTAL_HISTORI   OFF                                                                                                                                                      
 UA_ENABLE_HARDENING              ON                                                                                                                                                       
 UA_ENABLE_HISTORIZING            OFF                                                                                                                                                      
 UA_ENABLE_IMMUTABLE_NODES        OFF                                                                                                                                                      
 UA_ENABLE_JSON_ENCODING          OFF                                                                                                                                                      
 UA_ENABLE_MALLOC_SINGLETON       OFF                                                                                                                                                      
 UA_ENABLE_METHODCALLS            ON                                                                                                                                                       
 UA_ENABLE_MICRO_EMB_DEV_PROFIL   OFF                                                                                                                                                      
 UA_ENABLE_NODEMANAGEMENT         ON                                                                                                                                                       
 UA_ENABLE_NODESET_COMPILER_DES   ON                                                                                                                                                       
 UA_ENABLE_PUBSUB                 ON                                                                                                                                                       
 UA_ENABLE_PUBSUB_CUSTOM_PUBLIS   OFF                                                                                                                                                      
 UA_ENABLE_PUBSUB_DELTAFRAMES     OFF                                                                                                                                                      
 UA_ENABLE_PUBSUB_ETH_UADP        ON                                                                                                                                                       
 UA_ENABLE_PUBSUB_INFORMATIONMO   OFF                                                                                                                                                      
 UA_ENABLE_PUBSUB_INFORMATIONMO   OFF                                                                                                                                                      
 UA_ENABLE_PUBSUB_MQTT            ON                                                                                                                                                       
 UA_ENABLE_QUERY                  OFF                                                                                                                                                      
 UA_ENABLE_STATIC_ANALYZER        OFF                                                                                                                                                      
 UA_ENABLE_STATUSCODE_DESCRIPTI   ON                                                                                                                                                       
 UA_ENABLE_SUBSCRIPTIONS          ON                                                                                                                                                       
 UA_ENABLE_SUBSCRIPTIONS_ALARMS   OFF                                                                                                                                                      
 UA_ENABLE_SUBSCRIPTIONS_EVENTS   ON                                                                                                                                                       
 UA_ENABLE_TYPEDESCRIPTION        ON                                                                                                                                                       
[... cut ...]
 UA_ENABLE_WEBSOCKET_SERVER       OFF
```

I compiled server_ctt with the following command:

```bash
gcc -std=c99 open62541.c server_ctt.c -o server_ctt
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Raspbian OS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-01-25:
The first argument has to be the server certificate.

mikini, 2020-03-24:
I'm using the patch below to overcome this annoyance.

It changes the default state that the command line option "--enableAnonymous" also switches (which decides whether to remove the UA_USERTOKENTYPE_ANONYMOUS from allowed authentication methods) and makes it possible to execute the server without command line arguments in anonymous mode. This is nice for quick testing or experimentation without being required to mock about with certificates.

The parsing of the command line is a bit rudimentary and doesn't adhere to the announced usage (the .der cert is [indicated as being optional if UA_ENABLE_ENCRYPTION is not set](https://github.com/open62541/open62541/blob/master/examples/server_ctt.c#L927), but if any arguments are present [it has to be the first](https://github.com/open62541/open62541/blob/master/examples/server_ctt.c#L967)).

```
diff --git a/examples/server_ctt.c b/examples/server_ctt.c
index 15e051aa..846ce9ec 100644
--- a/examples/server_ctt.c
+++ b/examples/server_ctt.c
@@ -1009,7 +1009,7 @@ int main(int argc, char **argv) {
 
 #endif /* UA_ENABLE_ENCRYPTION */
 
-    UA_Boolean enableAnon = false;
+    UA_Boolean enableAnon = true;
     UA_Boolean enableTime = false;
 
     /* Loop over the remaining arguments */

```

andreasebner, 2025-02-04:
This issue seems to be outdated as the ctt example is not public for the latest releases. If there are known issues, feel free to open a new issue.