open62541/open62541 #3396: Add Historical_Data variables to DataBase

mojaxx, 2020-01-26:
Hi every body

I am running open62541 on BeagleBone Black and so far i run all the example on it.
But i have one question....
I want to make my BeagleBone server and store Historical_Data in a database on it and access this data on my computer as a client (for example by UA expert).
Does any one have an idea about how i can do this??

Thanks.

jpfr, 2020-01-26:
The current historicaldata-plugin only keeps data in-memory.
Feel free to make a database backend. 😄 
https://github.com/open62541/open62541/tree/master/plugins/historydata

mojaxx, 2020-01-27:
Thanks @jpfr

jpfr, 2020-01-27:
Not sure we're on the same wave-length here.
The backend of the historical data code is a plugin.

The database-based historical plugin does not yet exist.
But you can write one. Then you would replace the current in-memory backend with that in your server configuration.

mojaxx, 2020-01-28:
Hi @jpfr 
Thanks for your help.

closed 2020-01-28 by mojaxx