open62541/open62541 #3398: MQTT Publish with Json

waheedejaz, 2020-01-27:
## Minor Fix in tutorial_pubsub_mqtt_publish.c 
UA PubSub with MQTT example gives the option of JSON encoding but doesn't send JSON encoded. It always send UADP endcoded data.

## Background Information / Reproduction Steps
Build the solution with examples with PUBSUB and MQTT options. 
Run the example: `tutorial_pubsub_mqtt_publish --json`
Open another terminal and run: `mosquitto_sub -t customTopic`
Mosquitto clients needs to installed on the system. `sudo apt-get install mosquitto-clients` for ubuntu.

The mosquitto sub receives data but these are weird shapes, not json data.

`�e�M��Le�M�� �������
�^�M����^�M�� �������
�竵M����竵M�� �������
6;��M���:��M�� �������
6�D�M����D�M�� �������
2dM���	��M�� �������
`

OS: Ubunut 18
open62541 Release: latest on 27-Jan-2019 or the commit: ba060096bcec7335b812820c3e2a31a9033097bf

## Culprit Code

>Starting from line#130 

```
/* decide whether to use JSON or UADP encoding*/
#ifdef UA_ENABLE_JSON_ENCODING
    if(useJson)
        writerGroupConfig.encodingMimeType = UA_PUBSUB_ENCODING_JSON;
    else
#endif
        writerGroupConfig.encodingMimeType = UA_PUBSUB_ENCODING_UADP;
    writerGroupConfig.messageSettings.encoding             = UA_EXTENSIONOBJECT_DECODED;
    writerGroupConfig.messageSettings.content.decoded.type = &UA_TYPES[UA_TYPES_UADPWRITERGROUPMESSAGEDATATYPE];
    /* The configuration flags for the messages are encapsulated inside the
     * message- and transport settings extension objects. These extension
     * objects are defined by the standard. e.g.
     * UadpWriterGroupMessageDataType */
    UA_UadpWriterGroupMessageDataType *writerGroupMessage  = UA_UadpWriterGroupMessageDataType_new();
    /* Change message settings of writerGroup to send PublisherId,
     * WriterGroupId in GroupHeader and DataSetWriterId in PayloadHeader
     * of NetworkMessage */
    writerGroupMessage->networkMessageContentMask =
        (UA_UadpNetworkMessageContentMask)(UA_UADPNETWORKMESSAGECONTENTMASK_PUBLISHERID |
        (UA_UadpNetworkMessageContentMask)UA_UADPNETWORKMESSAGECONTENTMASK_GROUPHEADER |
        (UA_UadpNetworkMessageContentMask)UA_UADPNETWORKMESSAGECONTENTMASK_WRITERGROUPID |
        (UA_UadpNetworkMessageContentMask)UA_UADPNETWORKMESSAGECONTENTMASK_PAYLOADHEADER);
    writerGroupConfig.messageSettings.content.decoded.data = writerGroupMessage;
```

Even if _useJson_ variable is `true`, the code only changes the _encodedMimeType_ field is set to JSON encoding. The `decoded.type` field is always UADP as well as the contentMask flags.

Used CMake options:

```bash
cmake -DUA_BUILD_EXAMPLES -DUA_ENABLE_JSON_ENCODING -DUA_ENABLE_PUBSUB -DUA_ENABLE_PUBSUB_MQTT ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

waheedejaz, 2020-01-27:
## Solution

```
    /* decide whether to use JSON or UADP encoding*/
#ifdef UA_ENABLE_JSON_ENCODING
    if(useJson) {
        writerGroupConfig.encodingMimeType = UA_PUBSUB_ENCODING_JSON;
        writerGroupConfig.messageSettings.encoding             = UA_EXTENSIONOBJECT_DECODED;

		writerGroupConfig.messageSettings.content.decoded.type = &UA_TYPES[UA_TYPES_JSONDATASETWRITERMESSAGEDATATYPE];
		/* The configuration flags for the messages are encapsulated inside the
		 * message- and transport settings extension objects. These extension
		 * objects are defined by the standard. e.g.
		 * UadpWriterGroupMessageDataType */
		Json_writerGroupMessage = UA_JsonDataSetWriterMessageDataType_new();
		/* Change message settings of writerGroup to send PublisherId,
		 * WriterGroupId in GroupHeader and DataSetWriterId in PayloadHeader
		 * of NetworkMessage */
		Json_writerGroupMessage->dataSetMessageContentMask =
			(UA_JsonDataSetMessageContentMask)(UA_JSONDATASETMESSAGECONTENTMASK_DATASETWRITERID |
			(UA_JsonDataSetMessageContentMask)UA_JSONDATASETMESSAGECONTENTMASK_SEQUENCENUMBER |
			(UA_JsonDataSetMessageContentMask)UA_JSONDATASETMESSAGECONTENTMASK_TIMESTAMP |
			(UA_JsonDataSetMessageContentMask)UA_JSONDATASETMESSAGECONTENTMASK_STATUS);
		writerGroupConfig.messageSettings.content.decoded.data = Json_writerGroupMessage;
    }

    else
#endif
    {
		writerGroupConfig.encodingMimeType = UA_PUBSUB_ENCODING_UADP;
		writerGroupConfig.messageSettings.encoding             = UA_EXTENSIONOBJECT_DECODED;
		writerGroupConfig.messageSettings.content.decoded.type = &UA_TYPES[UA_TYPES_UADPWRITERGROUPMESSAGEDATATYPE];
		/* The configuration flags for the messages are encapsulated inside the
		 * message- and transport settings extension objects. These extension
		 * objects are defined by the standard. e.g.
		 * UadpWriterGroupMessageDataType */
		writerGroupMessage  = UA_UadpWriterGroupMessageDataType_new();
		/* Change message settings of writerGroup to send PublisherId,
		 * WriterGroupId in GroupHeader and DataSetWriterId in PayloadHeader
		 * of NetworkMessage */
		writerGroupMessage->networkMessageContentMask =
			(UA_UadpNetworkMessageContentMask)(UA_UADPNETWORKMESSAGECONTENTMASK_PUBLISHERID |
			(UA_UadpNetworkMessageContentMask)UA_UADPNETWORKMESSAGECONTENTMASK_GROUPHEADER |
			(UA_UadpNetworkMessageContentMask)UA_UADPNETWORKMESSAGECONTENTMASK_WRITERGROUPID |
			(UA_UadpNetworkMessageContentMask)UA_UADPNETWORKMESSAGECONTENTMASK_PAYLOADHEADER);
		writerGroupConfig.messageSettings.content.decoded.data = writerGroupMessage;
    }
```

and then at the end of the function:
```
    if (useJson)
    {
    	UA_JsonDataSetWriterMessageDataType_delete(Json_writerGroupMessage);
    }
    else
    {
    	UA_UadpWriterGroupMessageDataType_delete(writerGroupMessage);
    }
```
gives the following output:
>mosquitto_usb -t customTopic

> "{\"MessageId\":null,\"MessageType\":\"ua-data\",\"Messages\":[{\"DataSetWriterId\":62541,\"SequenceNumber\":30,\"MetaDataVersion\":{\"MajorVersion\":1996169766,\"MinorVersion\":1996169716},\"Timestamp\":\"2020-01-26T17:18:37.841621Z\",\"Payload\":{\"Server localtime\":{\"Type\":13,\"Body\":\"2020-01-26T17:18:37.841634Z\"}}}]}"

waheedejaz, 2020-02-04:
I noticed that I have used the incorrect datatype for WriterGroup message settings. I have used JsonDataSetWriter instead of JsonWriterGroupMessageDataType in WriterGroup Config. It was running successfully because the flags get set either way. But this needs to be corrected. Currently, JsonWriterGroupMessageDataType is not genertated automatically. tools/Schema/datatypes_pubsub.txt also needs to be updated. UA_Server_addWriterGroup() function also needs to validate the correct messageSettings type.
Creating another pull request in a moment and then we can close this issue.

jpfr, 2020-02-04:
Thanks for this @waheedejaz .
Still, I am pretty sure that the `--json` command has been working prior to #3410.
Provided that json encoding is enabled in the cmake settings...

Looking forward to the fix of the above.

closed 2020-02-06 by waheedejaz