open62541/open62541 #3401: Crash on UA_Server_delete

Maerdl, 2020-01-28:
Hi together,

while calling UA_Server_delete() my app running on an embedded system crashs.
I tracked down the issue to ```deleteNodeVisitor(NodeEntry *entry, void *data);```
The pointer entry has an ugly illegal address. And because of deleting this illegal address the system crashs.

Can yomebody provide me some hints how to detect who is adding such an entry?

Thanks a lot !

jpfr, 2020-01-28:
Hmm. Difficult.
Can you extract a small reproducer case that compiles on other systems as well?

Maerdl, 2020-01-28:
Hi @jpfr ,
thanks for your answer.
I tried to reproduce the crash on win32(msvcc) unfortunately I can not see the crash there.
At the moment i compare the two projects to ensure that they behave exact in the same manner.

Maerdl

Maerdl, 2020-01-29:
To keep you up to date, I tried the following things:

- 1.  I added a printf in ```deleteNodeVisitor``` to check the nodeid.
It always crashs on node id 0.
- 2. I build an win32 example. No crash there. I let it run in a dr. memory instance. And it seems that there is a leaked object. But to be honest, as I'm not that deep in dr. memory, I'm not sure if it is a flase positive. See the output:
```Error 1: LEAK 132 direct bytes 0x023ddcd8-0x023ddd5c + 58088 indirect bytes
 0 replace_calloc                               [d:\drmemory_package\common\alloc_replace.c:2616]
 1 newEntry                                     [c:\workcenter\opcua_open62541\open62541\plugins\ua_nodestore_default.c:101]
 2 UA_Nodestore_newNode                         [c:\workcenter\opcua_open62541\open62541\plugins\ua_nodestore_default.c:128]
 3 AddNode_raw                                  [c:\workcenter\opcua_open62541\open62541\src\server\ua_services_nodemanagement.c:867]
 4 Operation_addNode_begin                      [c:\workcenter\opcua_open62541\open62541\src\server\ua_services_nodemanagement.c:920]
 5 Operation_addNode                            [c:\workcenter\opcua_open62541\open62541\src\server\ua_services_nodemanagement.c:1247]
 6 __UA_Server_addNode                          [c:\workcenter\opcua_open62541\open62541\src\server\ua_services_nodemanagement.c:1306]
 7 UA_Server_addReferenceTypeNode               [c:\workcenter\opcua_open62541\open62541\include\open62541\server.h:1139]
 8 addReferenceTypeNode                         [c:\workcenter\opcua_open62541\open62541\src\server\ua_server_ns0.c:69]
 9 UA_Server_createNS0_base                     [c:\workcenter\opcua_open62541\open62541\src\server\ua_server_ns0.c:121]
10 UA_Server_initNS0                            [c:\workcenter\opcua_open62541\open62541\src\server\ua_server_ns0.c:752]
11 UA_Server_init                               [c:\workcenter\opcua_open62541\open62541\src\server\ua_server.c:258] ```

cochicde, 2020-01-29:
Hi,

which architecture are you using? Is it freeRTOS? Crashing in embedded but not in win32 or linux is many times a sympton that some memory is not set to zero after allocation. In this case my first thought is that the memory functions are not well, probably calloc is not zeroing the memory. That would be my first guess.

Also, I never used dr. memory, but it seems that that's not the whole output. 

The more information you give, the easier is to spot the error

Maerdl, 2020-01-29:
Hi @cochicde 
the architecture is ARMv7E-M the embedded os from SEGGER.
I did a quick check to the memory functions and they are all working as expected.
Regarding the stacktrace of dr. memory i assume that this is not my problem.

Unfortunately i can not test the full namespace as of its size.

Regards
Maerdl

cochicde, 2020-01-29:
does that mean that you're using embOS? did you code the the architecture for open62541 code for it? which commit are you using?

Maerdl, 2020-01-29:
@cochicde i ported the stack to be usable with embos.
I use the v1.0 branch with a cherry pick to be able to blacklist the full namespace.

What i see now is that if i build with a REDUCED Namespace there is no crash.
If i use my blacklisted FULL Namespace the crash is back. I use a blacklist to full namespace to have engineering units support.

Maerdl

Maerdl, 2020-02-03:
Any chance to check what nodes are missing ?

andreasebner, 2025-02-04:
This issue appears to be deprecated and the underlying code path has been reworked. If there are known issues, feel free to open a new issue.

closed 2025-02-04 by andreasebner