open62541/open62541 #3403: Network layer timeouts

andranik-s, 2020-01-29:
## Description

It seems to me that the timeout is not shared between network layers in `UA_Server_run_iterate`.
Basically I mean that this code:
https://github.com/open62541/open62541/blob/359f56e40dea85d69b1008184c1fd309779e1ac7/src/server/ua_server.c#L623-L630
should have `timeout = timeout / server->config.networkLayersSize` so that we don't miss next repeated callback if there are more than one layer.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-01-29:
This is fixed in #2408.

jpfr, 2021-12-16:
Done via the new EventLoop that is now incoporated into client+server+pubsub.

closed 2021-12-16 by jpfr