open62541/open62541 #3404: Two alarm conditions with the same source

CarlottaPat, 2020-01-29:
I'm using the open62541 library for my server development and I need to implement Alarms & Conditions in it.
I'm trying to create two alarm conditions with the same condition source object.
The call of UA_Server_triggerConditionEvent always fails when I pass the nodeid of the firstly created condition.
The returned error is UA_STATUSCODE_BADNOTFOUND.

Thank you for your help.

sqadasi, 2020-01-29:
Hello, this problem should've been fixed in #3373. Are you using the latest master?

CarlottaPat, 2020-01-29:
hanks for your reply!
I'm using the latest master.
However, if I create a source object and 2 exposed conditions with the statement

UA_Server_createCondition(server,
                          UA_NODEID_NULL,
                          conditionType,
                          uaConditionName,
                          conditionSourceObjectId,
                          UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT), // the condition is exposed
                          &conditionNodeId);

the status code returned by UA_Server_triggerConditionEvent for the first condition is UA_STATUSCODE_BADNOTFOUND.

CarlottaPat, 2020-01-29:
I have rebuilt the latest open62541 library and now it works!!!
I apologize for the time I made you lose. Thank you for your attention

CarlottaPat, 2020-01-29:
Just another question please.
If I try to acknowledge or confirm one of these alarms (for example by the UA Expert Client), it seems that both conditions are involved in this operation.
Both alarms have the same values for AckedState and for ConfirmedState.
What am I doing wrong?

sqadasi, 2020-01-29:
I can't tell without looking at your example. I tried to creat another condition, and they seem to have different values (as they suppose to have).
However, an error could occur (Events may get lost, and you won't be able to call the methods at all due to wrong event id) if the server and/or client don't have suitable event queue size. Your queue size should be at least equal to number of conditions + 2 (for refresh events).

CarlottaPat, 2020-01-30:
I have still some difficulties in commenting, acknowleding or confirming my two conditions that have the same source and are of the same type.
I have noticed that the addCommentMethodCallback, acknowledgeMethodCallback and confirmMethodCallback methods call getConditionBranchNodeId to find out the correct condition instance id.
They all call getConditionBranchNodeId passing the ConditionType and the correct condition id must be found through the lastEventId member. Could this cause problems?
Thanks in advance.

sqadasi, 2020-02-03:
It would be easier to help if we could see your implementation:)
As I said, I could not reproduce that error and the only thing I can think right now which could cause the problem, is that you lose some Events due to small event queue size.

CarlottaPat, 2020-02-03:
The source files of my OpcUaBaseAlarm class are attached to this comment. I'm using Qt 5.14.0.
To test this class I have simply inserted the following two instructions in my server code:
    OpcUaBaseAlarm* testAlarm = new OpcUaBaseAlarm(this, "A1TestAlarm", "GenericAlarms", true);
    OpcUaBaseAlarm* testAlarm2 = new OpcUaBaseAlarm(this, "A2TestAlarm", "GenericAlarms", true);

[OpcUaAlarm.zip](https://github.com/open62541/open62541/files/4148535/OpcUaAlarm.zip)

o6-nbeier, 2026-10-20:
We were not able to reproduce this. If you still encounter this issue, please open a new issue with up to date information.

Closing for now.

closed 2026-10-20 by o6-nbeier