open62541/open62541 #3412: How to generate own unique NodeId with desired namespace?

magiino, 2020-02-01:
## Description
I have noticed that automatic generated node ids are always in ns=0, even if the new added node is created from type which has node id with ns=2. How could I generate my own random unique node id with good namespace? Is there some ultility function for this in open62541? Or should I use the same code as in *UA_Nodestore_insertNode()* .. see picture below?? I prefer numeric node ids, but it can be of another type.

[image]

## Checklist

Please provide the following information:

 - [ x ] open62541 Version (release number or git tag): branch 1.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

sqadasi, 2020-02-03:
I guess generated unique NodeIds are always numeric. Just create a numeric NodeId with desired ns and identifier = 0, then pass it to the Node create function.
`NodeId nodeId = UA_NODEID_NUMERIC(desiredNameSpace, 0);`

jpfr, 2020-02-03:
Indeed. The nodestore will create a unique NodeId if the requested identifier is `i=0`.
The namespace is preserved.

However, the generated NodeId is not necessarily truly random.
This is in the choice of the nodestore plugin implementor.

See here for an example:
https://github.com/open62541/open62541/blob/5640a1372ebb9b27c2fbaa17be9c7fca93fae663/plugins/ua_nodestore_hashmap.c#L314

Another possibility is to use Guid identifiers for the NodeId.
We have `UA_Guid_random` to generate random ones.

Please reopen if questions remain.

closed 2020-02-03 by jpfr

magiino, 2020-02-05:
Thanks! I Can confirm that it works.

INPUT
`NodeId requestedNodeId = UA_NODEID_NUMERIC(desiredNameSpace, 0);`
`NodeId outNewNodeId = UA_NODEID_NUMERIC(0, 0);`

OUTPUT
`NodeId outNewNodeId` contains  `UA_NODEID_NUMERIC(desiredNameSpace, generated_numeric id);`