open62541/open62541 #3421: client_async on Windows don't call WSAStartup

nokem61, 2020-02-05:
## Description
Example - client_async does not work on Windows because WSAStartup is not called.

In function `UA_ClientConnectionTCP` is colled `UA_initialize_architecture_network();`
but `UA_Client_connect_async` use `UA_ClientConnectionTCP_init` where `UA_initialize_architecture_network();` is not called.

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): tag 1.0.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached [log.txt](https://github.com/open62541/open62541/files/4159300/log.txt)
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Soysauce007, 2020-03-19:
I suffered the same problem as you, have you solved it now?
It works very well when I use the _server.exe_, but it doesn't work when I use the _client.exe_

```
[2020-03-19 11:06:26.336 (UTC+0800)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-03-19 11:06:26.337 (UTC+0800)] warn/network       DNS lookup of localhost failed with error 应用程序没有调用 WSAStartup，或者 WSAStartup 失败。
```

Thanks

Matt-Edwards, 2020-04-24:
I'm having the same problem on Windows

Matt-Edwards, 2020-04-24:
I switched from master to v1.0.1 and now my client connects and runs.

timosmd, 2020-05-04:
I receive the same error message. Tried Branches "Master" and "0.1" - both didn't work...

jpfr, 2020-05-21:
This was fixed a view days ago.

jpfr, 2020-05-21:
Few days...

closed 2020-05-21 by jpfr