open62541/open62541 #3422: Measure OPC UA Performance

kusumy, 2020-02-05:
## Description
Hi everyone,
I have a assignment to measure OPC UA performance, such as:

a. Delay time, from server to client
b. Bytes allocated for each item/node/object sent by server to client

Since i am still new to OPC UA, i hope someone can give me a direction how to measure above parameters.

Regards

jpfr, 2020-02-07:
Hey @kusumy ,

for measuring the time delay, you could just use network dumps and Wireshark to extract timings.
Or you put instrumentation around a call to UA_Client_readValueAttribute or similar.

For the server case, that strongly depends on *what* you want to send.
The read service returns values from the information model.
The call service is basically a remote procedure call.
The browse service allows you to inspect the object structure of the information model in the server.

To learn about this,

- Read part 4 of the standard (available on the foundation page)
- Look at some OPC UA traffic in the Wireshark tool
- Break with a debugger in this method and look at how services are called and responses sent: https://github.com/open62541/open62541/blob/b9b194ff361e5a70db46c0aff37c334699ad6330/src/server/ua_server_binary.c#L506

kusumy, 2020-02-07:
Nice info.
I will definitely try it.

Thank you

jpfr, 2020-02-07:
You're welcome.
Please reopen if required.

closed 2020-02-07 by jpfr