open62541/open62541 #3430: open62541 as LDS-ME server

jdhanasekar, 2020-02-10:
Hi open6241 community,

I am new to open62541 OPC UA. I want to bring up a LDS-ME server. 
Will ```open62541``` support LDS-ME server?. If so, How can I use ```open62541``` as ```LDS-ME``` server ?

Please provide me your comments. It will be helpful for my project.

Thanks in Advance,

-Dhanasekar

Pro, 2020-02-12:
It's already fully supported. See the example here:
https://github.com/open62541/open62541/blob/master/examples/discovery/server_lds.c

closed 2020-02-12 by Pro

jdhanasekar, 2020-02-12:
I have enabled ```UA_ENABLE_AMALGAMATION``` and ```UA_ENABLE_DISCOVERY_MULTICAST``` and compiled in Linux system. 
Finally, I have got ```myServer``` executable file and when I run myserver, open62541 server is running. 
When I try to access open62541 server in Prosys Client I am getting below error,

```
[2020-02-12 19:50:35.321 (UTC+0550)] warn/server	Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2020-02-12 19:50:35.321 (UTC+0550)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-02-12 19:50:35.321 (UTC+0550)] info/network	TCP network layer listening on opc.tcp://quanta:4840/
[2020-02-12 19:51:14.406 (UTC+0550)] info/network	Connection 5 | New connection over TCP from 10.1.60.19
[2020-02-12 19:51:14.414 (UTC+0550)] info/channel	Creating a new SecureChannel
[2020-02-12 19:51:14.421 (UTC+0550)] info/channel	Connection 5 | SecureChannel 1 | Opened SecureChannel
[2020-02-12 19:51:14.434 (UTC+0550)] warn/channel	Connection 5 | SecureChannel 1 | The server sends no timestamp in the request header. See the 'verifyRequestTimestamp' setting.
[2020-02-12 19:51:14.459 (UTC+0550)] info/channel	Connection 5 | SecureChannel 1 | CloseSecureChannel
[2020-02-12 19:51:14.459 (UTC+0550)] info/network	Connection 5 | Closed

```

Pro, 2020-02-12:
`The server sends no timestamp in the request header. See the 'verifyRequestTimestamp' setting.` already tells you what to do...

You need to disable the timestamp verification in the server settings.

E.g. see here:
https://github.com/open62541/open62541/blob/e7d3e385e5bcec16a7967764490690879adde3f1/examples/server_ctt.c#L1230

jdhanasekar, 2020-02-13:
Can I comment out this line ```config.verifyRequestTimestamp = UA_RULEHANDLING_WARN; ``` ?

jdhanasekar, 2020-02-14:
I have added LDS code (https://github.com/open62541/open62541/blob/master/examples/discovery/server_lds.c) to myServer.c. When I compile, I am getting below error.

```
myServer.c: In function ‘main’:
myServer.c:38:22: error: ‘UA_ServerConfig_Discovery {aka struct <anonymous>}’ has no member named ‘mdns’
     config->discovery.mdns.mdnsServerName = UA_String_fromChars("LDS");
                      ^
myServer.c:48:22: error: ‘UA_ServerConfig_Discovery {aka struct <anonymous>}’ has no member named ‘mdns’
     config->discovery.mdns.serverCapabilitiesSize = 1;
                      ^
myServer.c:51:22: error: ‘UA_ServerConfig_Discovery {aka struct <anonymous>}’ has no member named ‘mdns’
     config->discovery.mdns.serverCapabilities = caps;

```

It seems that code inside this check ```#ifdef UA_ENABLE_DISCOVERY_MULTICAST``` in server_config.h is not getting executed. I think I need to enable this ```#UA_ENABLE_DISCOVERY_MULTICAST```.

Please help me to solve this error.