open62541/open62541 #3434: Nodeset Compiler for Siemens OPC UA Modeling Editor

prrw, 2020-02-12:
## Description
[image]

Building a model using **3DFrame** type from Structure in OPCUA Modeling tool of Siemens and then initializing the values to 0, leads to an error in Nodeset Compiler as described below. The header file generated from the XML using the Nodeset Compilor has the following code inside:
```
PS D:\local\open62541-master\tools\nodeset_compiler> python ./nodeset_compiler.py --types-array=UA_TYPES --existing D:/l
ocal/open62541-master/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml --xml D:/pkrawal/Desktop/frame.xml D:/pkrawal/Desktop/M
ultiPROM_Lasertracker
INFO:__main__:Preprocessing (existing) D:/local/open62541-master/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing D:/pkrawal/Desktop/frame.xml
Traceback (most recent call last):
  File "./nodeset_compiler.py", line 131, in <module>
    ns.addNodeSet(xmlfile, typesArray=getTypesArray(nsCount))
  File "D:\local\open62541-master\tools\nodeset_compiler\nodeset.py", line 211, in addNodeSet
    nodesets = dom.parseString(fileContent).getElementsByTagName("UANodeSet")
  File "D:\python27\lib\xml\dom\minidom.py", line 1928, in parseString
    return expatbuilder.parseString(string)
  File "D:\python27\lib\xml\dom\expatbuilder.py", line 940, in parseString
    return builder.parseString(string)
  File "D:\python27\lib\xml\dom\expatbuilder.py", line 223, in parseString
    parser.Parse(string, True)
xml.parsers.expat.ExpatError: not well-formed (invalid token): line 116, column 21
```
Line number 116 is basically this one:
[image]

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_AMALGAMATION=TRUE
```
How to solve this problem ? 

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):  https://github.com/open62541/open62541/commit/6574a46bd11d1c933930b2f638b9efd5ac4c10a0
 - [x] Other OPC UA SDKs used (client or server): Siemens OPC UA Modeling Editor 1.9.4
 - [x] Operating system:  Windows 10 with Microsoft Visual Studio 2015
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-02-13:
Can you run the xml with another xml validation (homepage)?
The error happens internally in the python library that parses the xml.

prrw, 2020-02-13:
@jpfr you are right. The XML validator shows an error too and in this line:
```
                    <3DFrame xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">
```
The error says, Line 116 Column 22 : The content of elements must consist of well-formed character data or markup.  Is it because ```3DFrame``` starts with a number ?

jpfr, 2020-02-13:
Yes. You cannot start names with numbers or the string "xml".

prrw, 2020-02-13:
okay. Got it. I changed ```3DFrame``` to ```ThreeDFrame``` in xml. And now i get this error:
```
INFO:__main__:Preprocessing (existing) D:/local/open62541-master/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing D:/pkrawal/Desktop/frame.xml
ERROR:datatypes:ns=1;i=1010: Expected ExtensionObject to hold a variable of type ns=0;3DFrame but found ThreeDFrame instead.
INFO:__main__:Generating Code for Backend: open62541
INFO:__main__:NodeSet generation code successfully printed
```
How should it be in the xml so that it can be compiled by the Nodeset Compilor ?

jpfr, 2020-02-13:
Apparently you still have a DataType `3DFrame` defined somewhere.
Is this part of some Companion Specification?

prrw, 2020-02-14:
@jpfr So 3DVector, 3DFrame, 3DCartesianCoordinates and 3DOrientation are defined in the document OPC 10001-11 which are amendments made to OPC 10000-5 which is the Information Model.
If you have a look at ```Opc.Ua.NodeSet2.xml```, 3DFrame is also defined there. OPC Foundation already knows about this wrong xml problem and thats why they have defined ```SymbolicName``` along with ```BrowseName``` for all data types which start with a number. 

Look at the example below. I compare a normal datatype and3DFrame defined in the ```Opc.Ua.NodeSet2.xml```

```
<UADataType NodeId="i=18814" BrowseName="3DFrame" SymbolicName="ThreeDFrame">
```
```
  <UADataType NodeId="i=12" BrowseName="String">
```

I assume that this means that, if Symbolic name is available, then symbolic name needs to be used in the XML generated. So there is a bug in Siemens Modeling Editor. But on the otherside, Nodeset Compiler should also interpret ```ThreeDFrame``` as ```3DFrame```. This must be changed. 
What are your thoughts ?

jpfr, 2020-02-14:
We have to resolve with the OPC Foundation.
Do you have access to the mantis bug tracker on their end?

prrw, 2020-02-14:
unfortunately no. Yes, this needs to be resolved.

o6-nbeier, 2026-10-20:
Thank you for taking the time to open this issue and describe it so clearly.

The problem has since been fixed on the maintained branches, so we are closing it as resolved.

closed 2026-10-20 by o6-nbeier