open62541/open62541 #3435: Question: Namespace Zero Full Size?

Mosbeard, 2020-02-13:
Hello,

in our project we want to implement an open62541 OPC UA Server on an embedded ESP32 microcontroller with a little custom namespace which depends on namespace DI and full namespace zero. Does someone know how much RAM and ROM we will need at least, when working with full namespace zero and namespace DI?

jpfr, 2020-02-13:
For full you will need 2-3MB.
But this will get massively better once PR #3411 has landed.

Mosbeard, 2020-02-14:
Thanks for information.
Do you mean 2-3 MB RAM?
And after this PR it is possible to hold the nodestore in ROM.

jpfr, 2020-02-14:
Yes.

saperlot, 2020-02-14:
With namespace minimal, i've got about 200kb code size and 50kb Read Only and 4kb Read Write Data. To give you an idea..

NoelGraf, 2021-05-28:
@Mosbeard can the issue be closed?

closed 2021-06-01 by NoelGraf